khoih-prog/WebSockets_Generic #1: Support as Http server and the webscok server on Arduino DUE

jakespeed1311, 2020-05-23:
Hello,
Sorry i want so talk about Websockets Server Examples. 

i am want wo work with the library with Arduino DUE and PlatformIO last version from Today.
My projectis based on AJAX now i want to switch some functions to Websockets.
I have an working Webserver with html files on SDCard, and i want to bring these Library as websock server on the DUE.  Normal Ethernet shiled with the WizNet 5100 controller and SDslot.

Is this posible ?
The PC ist only the client and the DUE is the Server for http and Websock paralel. 
The Due sends the data to the Client Browser on the pc.

I havent seen an Examble as Server also not found realy the DUE as suppotert, well only the list for the toolchain.

tanks
Jake

khoih-prog, 2020-05-24:
@jakespeed1311 

Thanks for your interest in the library.

Currently, SAM DUE is not supported. The WebSockets Server for many other boards is still not ready.

I'm in the process of porting the WebSockets library to **Teensy, SAM DUE and STM32** running with **W5x00, ENC28J60 and WiFiNINA**, as well as make **WebSockets Server** working in those supported boards.

I think by next week, the new version will have those features added. But I can't promise everything is working as planned.

The library will **hopefully** provide the WebSocket Server feature for the SAM DUE, but you have to write the application to use the feature for your purpose (SD Card, HTTP Server, etc.).

PS: If you'd like, you can write in your language, and I'll try to use Translator to convert it to English.

jakespeed1311, 2020-05-24:
Hello,
thank you for the good news, i hope for good things to come. The Http server and the SDCard output are already running. Now add the websock server, that would be really cool.
I keep my fingers crossed for a smooth adaptation to the new hardware
thanks
Jake

khoih-prog, 2020-05-24:
Some preliminary good news for SAM DUE, Teensy and STM32F

The WebSocket Server example has just been compiled smoothly, but it needs some important mods in the Arduino hardware package.

Still need some more tests to see if anything is broken. **If you're willing to test,** I'll post some fixes for you here early next week to have something to start working on.

khoih-prog, 2020-05-26:
Hi @jakespeed1311 

Good news. Just posted the new release v2.2.2

#### New in v2.2.2

1. Add support to ***Teensy*** boards, such as ***Teensy 4.1, 4.0. 3.6, 3.5, 3.2/3.1, 3.0, LC.***.
2. Add support to ***STM32 (Nucleo-144, Nucleo-64, Nucleo-32, Discovery, STM32F1, STM32F3, STM32F4, STM32H7, STM32L0, etc.)***.
3. Add support to ***SAM DUE***
4. Add WebSocketServer examples to those supported boards.

Remember to follows instructions in

[Important Notes to add support to SAM DUE](https://github.com/khoih-prog/WebSockets_Generic#important-notes)

The [Generic_WebSocketServerAllFunctionsDemo_W5500 example](examples/Generic/W5500/Generic_WebSocketServerAllFunctionsDemo_W5500) will show you how to run both EthernetWebServer and WebSockets Server for Generic boards (including SAM DUE) with W5x00 Ethernet.

Please post your test results if OK.

Good luck,

closed 2020-05-26 by khoih-prog

jakespeed1311, 2020-05-27:
Hello,

Tanks for the good news.

I will make test as soon als i can.

regards

Jake

 
 

Gesendet: Dienstag, 26. Mai 2020 um 05:43 Uhr

 

	Add support to SAM DUE

The Generic_WebSocketServerAllFunctionsDemo_W5500 example will show you how to run both EthernetWebServer and WebSockets Server for Generic boards (including SAM DUE) with W5x00 Ethernet.

jakespeed1311, 2020-05-30:
Hello,

i hope i can reach you on this way.  My email Jakespeed@web.de

 

 

1. i Use the normal http server from Arduino, you are using an better i know but i must cange all my work. Do the websock work with the standart arduino lib ?

 

2. I cant compile the Generic_WebsocketServer_W5500  demo  from Ver. 2.2.2 i uses Arduino IDE 1.8.12

    I havent checked the library in function so i wondering about the warnigns

3. I work with the latest ethernet libraty   2.0  because i work W5500 and W5100 cards

 

Please give me an hint.

Jake

 

 

 

+++++++++ begin output

 

 

Arduino: 1.8.12 (Windows 10), Board: "Arduino Due (Programming Port)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\admin\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Nextcloud\Eisenbahn_Fremo\arduino_Work\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10812 -build-path C:\Users\admin\AppData\Local\Temp\arduino_build_359102 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Nextcloud\Eisenbahn_Fremo\arduino_Work\libraries\WebSockets_Generic-master\examples\Generic\W5500\Generic_WebSocketServer_W5500\Generic_WebSocketServer_W5500.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\admin\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Nextcloud\Eisenbahn_Fremo\arduino_Work\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10812 -build-path C:\Users\admin\AppData\Local\Temp\arduino_build_359102 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Nextcloud\Eisenbahn_Fremo\arduino_Work\libraries\WebSockets_Generic-master\examples\Generic\W5500\Generic_WebSocketServer_W5500\Generic_WebSocketServer_W5500.ino
Using board 'arduino_due_x_dbg' from platform in folder: C:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Using core 'arduino' from platform in folder: C:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Detecting libraries used...
"C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10812 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\admin\\AppData\\Local\\Temp\\arduino_build_359102\\sketch\\Generic_WebSocketServer_W5500.ino.cpp" -o nul
Alternatives for WebSocketsServer_Generic.h: [WebSockets_Generic-master@2.2.2]
ResolveLibrary(WebSocketsServer_Generic.h)
  -> candidates: [WebSockets_Generic-master@2.2.2]
"C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10812 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_ [... cut ...] ude/" "-IC:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino\\avr\\dtostrf.c" -o "C:\\Users\\admin\\AppData\\Loca

jakespeed1311, 2020-05-30:
hi,

@khoih-prog

I found one mistake from me i overseen to change the due platform.txt cange. Sorry.

Now i try to test.

tx

Jake

 

Gesendet: Dienstag, 26. Mai 2020 um 05:43 Uhr

khoih-prog, 2020-05-30:
Hope you're OK now. This is my compile result of the same example for SAM DUE

```
In file included from /home/kh/Arduino/libraries/WebSockets_Generic-2.2.2/src/WebSocketsServer_Generic.h:43:0,
                 from /home/kh/Arduino/khoih-prog_working/WebSockets_Generic-2.2.2/examples/Generic/W5500/Generic_WebSocketServer_W5500/Generic_WebSocketServer_W5500.ino:26:
/home/kh/Arduino/libraries/WebSockets_Generic-2.2.2/src/WebSockets_Generic.h:59:6: warning: #warning Undef min/max in WebSockets_Generic [-Wcpp]
     #warning Undef min/max in WebSockets_Generic

      ^
In file included from /home/kh/Arduino/libraries/WebSockets_Generic-2.2.2/src/WebSocketsServer_Generic.h:43:0,
                 from /home/kh/Arduino/khoih-prog_working/WebSockets_Generic-2.2.2/examples/Generic/W5500/Generic_WebSocketServer_W5500/Generic_WebSocketServer_W5500.ino:26:
/home/kh/Arduino/libraries/WebSockets_Generic-2.2.2/src/WebSockets_Generic.h:180:4: warning: #warning Use SAM DUE in WebSockets_Generic [-Wcpp]
   #warning Use SAM DUE in WebSockets_Generic

    ^
In file included from /home/kh/Arduino/libraries/WebSockets_Generic-2.2.2/src/WebSockets_Generic.h:527:0,
                 from /home/kh/Arduino/libraries/WebSockets_Generic-2.2.2/src/WebSocketsServer_Generic.h:43,
                 from /home/kh/Arduino/khoih-prog_working/WebSockets_Generic-2.2.2/examples/Generic/W5500/Generic_WebSocketServer_W5500/Generic_WebSocketServer_W5500.ino:26:
/home/kh/Arduino/libraries/WebSockets_Generic-2.2.2/src/WebSockets_Generic-Impl.h:53:6: warning: #warning No CORE_HAS_LIBB64 [-Wcpp]
     #warning No CORE_HAS_LIBB64

      ^
In file included from /home/kh/Arduino/libraries/Ethernet/src/Ethernet.cpp:48:0:
/home/kh/Arduino/libraries/Ethernet/src/utility/w5100.h:68:2: warning: #warning Use 25MHz clock for W5200/W5500. Not for W5100 [-Wcpp]
 #warning Use 25MHz clock for W5200/W5500. Not for W5100
  ^
/home/kh/Arduino/libraries/Ethernet/src/utility/w5100.h:502:2: warning: #warning Use SAM3 architecture [-Wcpp]
 #warning Use SAM3 architecture
  ^
...

/home/kh/Arduino/libraries/Ethernet/src/utility/w5100.cpp:124:2: warning: #warning w5100.cpp Use fallback, change SS_PIN_DEFAULT to 10 [-Wcpp]
 #warning w5100.cpp Use fallback, change SS_PIN_DEFAULT to 10
  ^
Sketch uses 99912 bytes (19%) of program storage space. Maximum is 524288 bytes.

```

jakespeed1311, 2020-06-06:
Hello,
I hope you're okay. I was able to compile the webserver and it works.
Unfortunately I can't make the Generic_WebSocketServerAllFunctionsDemo_W5500.ino
to get it running. The web page is displayed, but I cannot open a websock with Chrome/Firefox.
Maybe you can test it again - something I am doing wrong. Unfortunately I can't get any further. 

Tanks for your help.
lg
Jake

Serial Monitor:
19:45:48.647 -> 0 Connected websocket clients ping: 1
19:45:49.004 -> 
19:45:49.004 -> Start Generic_WebSocketServerAllFunctionsDemo_W5500
19:45:49.004 -> [SETUP] BOOT WAIT 4
19:45:49.988 -> [SETUP] BOOT WAIT 3
19:45:51.005 -> [SETUP] BOOT WAIT 2
19:45:52.000 -> [SETUP] BOOT WAIT 1
19:45:53.574 -> 
19:45:53.574 -> W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 10
19:45:53.574 -> W5100::init: W5100, SSIZE =4096
19:45:53.574 -> WebSockets Server IP address: 192.168.0.111
19:45:58.533 -> [WS] 4 [handleHeader] Header read fin.
19:45:58.533 -> [ETHERNET_WEBSERVER] request handler not found
19:45:58.979 -> 0 Connected websocket clients ping: 1

I get this error in my browser:

(index):1 WebSocket connection to 'ws://192.168.0.111:81/' failed: Error during WebSocket handshake: Unexpected response code: 400
(anonymous) @ (index):1
(index):1 WebSocket Error  
Event
bubbles: false
cancelBubble: false
cancelable: false
composed: false
currentTarget: WebSocket {url: "ws://192.168.0.111:81/", readyState: 3, bufferedAmount: 0, onopen: ƒ, onerror: ƒ, ...}
defaultPrevented: false
eventPhase: 0
isTrusted: true
path: []
returnValue: true
srcElement: WebSocket {url: "ws://192.168.0.111:81/", readyState: 3, bufferedAmount: 0, onopen: ƒ, onerror: ƒ, ...}
target: WebSocket {url: "ws://192.168.0.111:81/", readyState: 3, bufferedAmount: 0, onopen: ƒ, onerror: ƒ, ...}
timeStamp: 89.90000002086163
type: "error"
__proto__: Event

khoih-prog, 2020-06-06:
Hi @jakespeed1311 

The WebSockets Server currently has severe bug, and currently is not working.

I think you got the same issue as the other 2 issues

1. [No free space new client / Counting of clients wrong?](https://github.com/khoih-prog/WebSockets_Generic/issues/2)
2. [ Major issue with WiFiNINA - Arduino IoT 33](https://github.com/khoih-prog/WebSockets_Generic/issues/3)

I'm working on that now. But the outcome and time frame to fix is not determined, and can be quite long. If you can find another quicker solution, I suggest you go with that.

jakespeed1311, 2020-06-06:
Hi,

ok, tanks. I am not shure iam not know the protocol in deep. But i will try my best i want to use the server on the DUE.

if i find out i let you know.

Greedings 

Jake

 

 

Gesendet: Samstag, 06. Juni 2020 um 21:34 Uhr

jakespeed1311, 2020-06-08:
Hello,

i checked up the lib and i mean that the client always get an new number.

So the filling of the array is wrong.

Bu i am not able to fix. I mean that in handle new clients should be an part like this to keep the client number.

 

void WebSocketsServer::handleNewClients(void) 
{
#if (WEBSOCKETS_NETWORK_TYPE ==  NETWORK_W5100)
while( client.available());
{
#endif

...

but there is no client.available function as it in ethernetclient is definded. 

 

Yes i have to learn that but today i have no change to fix this. if this is the problem.

 

this is the debug output ans some prints from mee to understand the program.

An if i am right after the [WS] is the client number, but this must be static.

Tanks

Jake

 

10:59:51.152 -> 
10:59:51.152 -> Start Generic_WebSocketServerAllFunctionsDemo_W5500
10:59:51.152 -> [SETUP] BOOT WAIT 4
10:59:52.150 -> [SETUP] BOOT WAIT 3
10:59:53.148 -> [SETUP] BOOT WAIT 2
10:59:54.135 -> [SETUP] BOOT WAIT 1
10:59:55.729 -> 
10:59:55.729 -> W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 10
10:59:55.729 -> W5100::init: W5100, SSIZE =4096
10:59:55.729 -> WebSockets Server IP address: 192.168.0.111
10:59:59.670 -> [WS] 0 [handleHeader] Header Komplett RX: GET / HTTP/1.1
10:59:59.670 -> get ist da
10:59:59.670 -> [WS] 0  index: 3
10:59:59.670 -> -------
10:59:59.670 -> [WS] 0  cURL:  
10:59:59.670 -> 3
10:59:59.670 -> [WS] 0  cURL: /
10:59:59.670 -> [WS] 0  URL: GET / HTTP/1.1
10:59:59.670 -> [WS] 0 [handleHeader] Header Komplett RX: Host: 192.168.0.111:91
10:59:59.704 -> [WS] 1 [handleHeader] Header Komplett RX: Connection: Upgrade
10:59:59.704 -> connection
10:59:59.704 -> upgrade
10:59:59.704 -> [WS] 1  cIsUpgrade : 1
10:59:59.704 -> [WS] 0 [handleHeader] Header Komplett RX: Pragma: no-cache
10:59:59.704 -> [WS] 1 [handleHeader] Header Komplett RX: Cache-Control: no-cache
10:59:59.704 -> [WS] 2 [handleHeader] Header Komplett RX: User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/83.0.4103.97 Safari/537.36
10:59:59.704 -> [WS] 0 [handleHeader] Header Komplett RX: Upgrade: websocket
10:59:59.739 -> upgrade
10:59:59.739 -> websocket
10:59:59.739 -> is socket
10:59:59.739 -> [WS] 1 [handleHeader] Header Komplett RX: Origin: http://192.168.0.111
10:59:59.739 -> [WS] 2 [handleHeader] Header Komplett RX: Sec-WebSocket-Version: 13
10:59:59.739 -> version
10:59:59.739 -> gwebsocket
10:59:59.739 -> [WS] 3 [handleHeader] Header Komplett RX: Accept-Encoding: gzip, deflate
10:59:59.739 -> [WS] 0 [handleHeader] Header Komplett RX: Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
10:59:59.774 -> [WS] 1 [handleHeader] Header Komplett RX: Sec-WebSocket-Key: 8PJC8Z75BzzAiwNWzCa6Aw==
10:59:59.774 -> cKey da
10:59:59.774 -> [WS] 2 [handleHeader] Header Komplett RX: Sec-WebSocket-Extensions: permessage-deflate; client_max_window_bits
10:59:59.774 -> headerValue
10:59:59.774 -> [WS] 3 [handleHeader] Header Komplett RX: Sec-WebSocket-Protocol: arduino
10:59:59.774 -> headerValue
10:59:59.774 -> [WS] 4 [handleHeader] Header read fin.
10:59:59.774 -> [WS] 4 [handleHeader]   - cURL: 
10:59:59.774 -> [WS] 4 [handleHeader]   - cIsUpgrade: 0
10:59:59.808 -> [WS] 4 [handleHeader]   - cIsWebsocket: 0
10:59:59.808 -> [WS] 4 [handleHeader]   - cKey: 
10:59:59.808 -> [WS] 4 [handleHeader]   - cProtocol: 
10:59:59.808 -> [WS] 4 [handleHeader]   - cExtensions: 
10:59:59.808 -> [WS] 4 [handleHeader]   - cVersion: 0
10:59:59.808 -> [WS] 4 [handleHeader]   - base64Authorization: 
10:59:59.808 -> [WS] 4 [handleHeader]   - cHttpHeadersValid: 0
10:59:59.808 -> [WS] 4 [handleHeader]   - cMandatoryHeadersCount: 0
10:59:59.840 -> 0
10:59:59.840 -> [handleHeader] Websocket connection incoming.
10:59:59.840 -> shakey:258EAFA5-E914-47DA-95CA-C5AB0DC85B11
10:59:59.840 -> runCbEvent(

 

Gesendet: Samstag, 06. Juni 2020 um 21:34 Uhr

khoih-prog, 2020-06-08:
Hi @jakespeed1311 

Thanks for your help. The same issue also happens while using other Ethernet/WiFi such as WiFiNINA, etc.
I'm narrow it down to the same **`client number`** issue, but still haven't had a solution yet.
Any more help will be appreciated.

jakespeed1311, 2020-06-27:
Hello,
I'm trying to fix it for the due, but unfortunately I don't understand what exactly is going wrong.
I was thinking about asking Paul Stoffregen for help, he is the one who developed the Ethernet lib.
The lib should always have the same client ID.
But like I said I'm just learning the whole thing.
If you should get further, that would be great, I can test it anyway.
greetings
Jake

 
 

Gesendet: Montag, 08. Juni 2020 um 16:24 Uhr

khoih-prog, 2020-07-16:
Hi @jakespeed1311 

Please try the new [WebSockets2_Generic Library](https://github.com/khoih-prog/WebSockets2_Generic) I believe there is no WebSocket Server issue as this [WebSockets_Generic Library](https://github.com/khoih-prog/WebSockets_Generic).

This is terminal debug output when running [SAMDUE-Ethernet-Server](examples/Generic/Ethernet/SAMDUE/SAMDUE-Ethernet-Server) on SAM DUE board with Ethernet W5100 shield using EthernetLarge library, and receiving WebSockets connection from an ESP8266 board.

```
Starting WebSockets2_Generic SAMDUE-Ethernet-Server on SAM DUE
[SETUP] BOOT WAIT 4
[SETUP] BOOT WAIT 3
[SETUP] BOOT WAIT 2
[SETUP] BOOT WAIT 1

W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 10
W5100::init: W5100, SSIZE =4096
WebSockets Server Running and Ready on SAM DUE
IP address: 192.168.2.153, Port: 8080
H[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.153
[WS] WebsocketsServer::recvHandshakeRequest: value = ********
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from ESP8266
[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.153
[WS] WebsocketsServer::recvHandshakeRequest: value = ********
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from ESP8266
HHHHHHHHH
```

### Why do we need this [WebSockets2_Generic library](https://github.com/khoih-prog/WebSockets2_Generic)

Many Web services require WebSockets library, which is so far written only for ESP8266/ESP32 boards. The ESP boards so far rely on the [Markus Sattler's WebSockets Library](https://github.com/Links2004/arduinoWebSockets) to connect to Alexa via Sinric or SinricPro skills.

This [WebSockets2_Generic library](https://github.com/khoih-prog/WebSockets2_Generic) is based on and modified from [Gil Maimon's ArduinoWebSockets Library](https://github.com/gilmaimon/ArduinoWebsockets) to provide support to many more boards, such as ***Arduino SAMD21, Adafruit SAMD21/SAMD51, nRF52, STM32, Teensy, SAM DUE, etc.*** and enable those boards to use WebSockets services, including voice-control Alexa along with Blynk. Those supported boards can also run ***WebSockets Server.*** The WebSockets can be used with ***ESP’s WiFi, WiFiNINA, W5x00/ENC28J60/LAN8742A Ethernet, ESP8266/ESP32-AT modules/shields.***

---

#### New in v1.0.1

1. Add support to ***SAM DUE*** boards.
2. Add support to ***Ethernet W5x00, using either Ethernet, EthernetLarge or Ethernet2 library.***
3. Restructure code to provide flexibility to support many more WiFi/Ethernet modules/shields in the future. Please delete the *.cpp files, replaced by *.hpp files, in the src directory, if *.cpp files still exist after installing new version.

#### New in v1.0.0

1. Add support to ***nRF52*** boards, such as ***AdaFruit Feather nRF52832, nRF52840 Express, BlueFruit Sense, Itsy-Bitsy nRF52840 Express, Metro nRF52840 Express, NINA_B302_ublox, NINA_B112_ublox, etc.***.
2. Add support to ***SAMD51 (Itsy-Bitsy M4, Metro M4, Grand Central M4, Feather M4 Express, etc.)***.
3. Add support to ***SAMD21 (ZERO, MKR, NANO_33_IOT, M0, M0 Pro, AdaFruit CIRCUITPLAYGROUND_EXPRESS, etc.)***.
4. Add Alexa / SINRIC examples and supports to those boards as well as ESP8266/ESP32

jakespeed1311, 2020-07-26:
Hello,

i am trying to modify the Examble Generic_WebSocketSeverAllFunctionsDemo_W550  to run with the Websockets2 library.

But i missed something so i dont get it wo work.

 

I have an Webserver running with he website that starts the websocket. And the websockserver that is sending data to the wsclient that was opend from the Website.

But i can't get it wo work.

Can you please help with this examble.

tx

Jake

 

 

 
 

Gesendet: Donnerstag, 16. Juli 2020 um 15:14 Uhr

 

This is terminal debug output when running SAMDUE-Ethernet-Server on SAM DUE board with Ethernet W5100 shield using EthernetLarge library, and receiving WebSockets connection from an ESP8266 board.

khoih-prog, 2020-07-27:
You can start from this modified sketch for SAM-DUE and [WebSockets2_Generic Library](https://github.com/khoih-prog/WebSockets2_Generic).

You have to spend more time to improve it. Currently the modifications are just to show you how to make it working basically.

I'm sorry I can't spend more time to solve your issue.

1) File

```
#include "defines.h"

#include <WebSockets2_Generic.h>

#include <EthernetWebServer.h>

using namespace websockets2_generic;

WebsocketsServer SocketsServer;

#define WEBSOCKETS_PORT     8080

// Change pins according to your boards
#define RED_LED     7
#define GREEN_LED   6
#define BLUE_LED    5

EthernetWebServer server(80);

#if 0
void webSocketEvent(uint8_t num, WStype_t type, uint8_t * payload, size_t length)
{
  switch (type)
  {
    case WStype_DISCONNECTED:
      //Serial.println( "[" + String(num) + "] Disconnected!");
      break;
    case WStype_CONNECTED:
      {
        //IPAddress ip = webSocket.remoteIP(num);
        //Serial.printf("[%u] Connected from %d.%d.%d.%d url: %s\n", num, ip[0], ip[1], ip[2], ip[3], payload);

        // send message to client
        webSocket.sendTXT(num, "Connected");
      }
      break;
    case WStype_TEXT:
      Serial.println( "[" + String(num) + "] get Text: " + String((char *) payload));

      if (payload[0] == '#')
      {
        // we get RGB data

        // decode rgb data
        uint32_t rgb = (uint32_t) strtol((const char *) &payload[1], NULL, 16);

        analogWrite(RED_LED, ((rgb >> 16) & 0xFF));
        analogWrite(GREEN_LED, ((rgb >> 8) & 0xFF));
        analogWrite(BLUE_LED, ((rgb >> 0) & 0xFF));
      }
      break;

    default:
      break;
  }
}
#endif

void setup()
{
#if (USE_ETHERNET_LIB || USE_ETHERNET2_LIB || USE_ETHERNET_LARGE_LIB)
[... 212 lines cut ...]
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from SAMD_NANO_33_IOT
[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.95
[WS] WebsocketsServer::recvHandshakeRequest: value = MDEyMzQ1Njc4OWFiY2RlZg==
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from NUCLEO_F767ZI
[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.95
[WS] WebsocketsServer::recvHandshakeRequest: value = MDEyMzQ1Njc4OWFiY2RlZg==
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from SAMD_NANO_33_IOT

```

jakespeed1311, 2020-08-04:
Hello,

i tested now, sorry for the delay. But i don't get it working complete.

First the websock on port 8080 was resonding, i test only with chrome   ip:8080 so no real client but an output on the Terminal.

 

The http server was not responding, i checked agin the ethernet2 lib and Ardunino mods and there are well.

Then i tryed the EterhnetWebServer example to see if its working, and yes that do.

 

I chekced the ip fixed and the dhcp.. but no luck with the Http screen.

The Server is not responding, i can't see that he will send data.

I only modified the ip in defines.h  and have a w5100 sheld.

 

 

Then i removed the ws server config and i see the html site.

So now both work but not togehter.

Ethernetwebserver 1.0.11

Websock2              1.05

Eternet2 + patcg   2.0

I run Arduino IDE 1.8.12

I used the lib/packets mods from Websock2 

 

 

Tanks for your help.

greedings

Jake

 

 

 

 
 

Gesendet: Montag, 27. Juli 2020 um 20:34 Uhr

khoih-prog, 2020-08-05:
I'm testing here and seeing both EthernetWebServer and WebSocketsServer working simultaneously.

Try this code and see

```
#include "defines.h"

#include <WebSockets2_Generic.h>

#include <EthernetWebServer.h>

using namespace websockets2_generic;

WebsocketsServer SocketsServer;

#define WEBSOCKETS_PORT     8080

EthernetWebServer server(80);

int reqCount = 0;                // number of requests received

void handleRoot()
{
#define BUFFER_SIZE     400
  
  char temp[BUFFER_SIZE];
  int sec = millis() / 1000;
  int min = sec / 60;
  int hr = min / 60;
  int day = hr / 24;

  snprintf(temp, BUFFER_SIZE - 1,
           "<html>\
<head>\
<meta http-equiv='refresh' content='5'/>\
<title>AdvancedWebServer %s</title>\
<style>\
body { background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }\
</style>\
</head>\
<body>\
<h2>Hi from EthernetWebServer!</h2>\
<h3>on %s</h3>\
<p>Uptime: %d d %02d:%02d:%02d</p>\
<img src=\"/test.svg\" />\
</body>\
</html>", BOARD_NAME, BOARD_NAME, day, hr % 24, min % 60, sec % 60);

  server.send(200, "text/html", temp);
}

void handleNotFound()
{
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";

  for (uint8_t i = 0; i < server.args(); i++)
  {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }

  server.send(404, "text/plain", message);
}

[... 134 lines cut ...]
W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 10
W5100::init: W5100, SSIZE =4096
WebSockets Server IP address: 192.168.2.93
WebSockets Server Running and Ready on SAM DUE
IP address: 192.168.2.93, Port: 8080
[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.93
[WS] WebsocketsServer::recvHandshakeRequest: value = MDEyMzQ1Njc4OWFiY2RlZg==
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from ESP8266
[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.93
[WS] WebsocketsServer::recvHandshakeRequest: value = MDEyMzQ1Njc4OWFiY2RlZg==
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from ESP8266

```

khoih-prog, 2020-08-05:
[image]

khoih-prog, 2020-08-05:
Running more than 8hrs continuously and both EthernetServer and WebSocketsServer are still OK

[image]

jakespeed1311, 2020-08-05:
Hi,

today i try everithing to fix the problem. But no luck.

I download and patched all, i read the manuals again. Http/ and WS Server.

Again the webserver demo is running fine. The Websock do respont test from an Chrome plugin.

Is there an option to set in http server ? To get both working.

I'm kind of desperate now.

 

16:29:22.385 ->
16:29:22.385 -> Starting WebSockets2_Generic SAMDUE-Ethernet_ServerAllFunctionsDemo on NMCSAM DUE
16:29:22.385 -> Ethernet using W5x00 and Ethernet Library
16:29:22.432 -> [SETUP] BOOT WAIT 4
16:29:23.423 -> [SETUP] BOOT WAIT 3
16:29:24.407 -> [SETUP] BOOT WAIT 2
16:29:25.423 -> [SETUP] BOOT WAIT 1
16:29:26.982 ->
16:29:26.982 -> W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 10
16:29:26.982 -> W5100::init: W5500, SSIZE =4096
16:29:26.982 -> WebSockets Server IP address: 192.168.2.95
16:29:26.982 -> WebSockets Server Running and Ready on SAM DUE
16:29:26.982 -> IP address: 192.168.2.95, Port: 8080
16:31:48.306 -> [WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.95:8080
16:31:48.306 -> [WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
16:31:48.306 -> [WS] WebsocketsServer::recvHandshakeRequest: value = no-cache
16:31:48.340 -> [WS] WebsocketsServer::recvHandshakeRequest: value = no-cache
16:31:48.340 -> [WS] WebsocketsServer::recvHandshakeRequest: value = Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/84.0.4147.105 Safari/537.36
16:31:48.340 -> [WS] WebsocketsServer::recvHandshakeRequest: value = websocket
16:31:48.340 -> [WS] WebsocketsServer::recvHandshakeRequest: value = chrome-extension://mdmlhchldhfnfnkfmljgeinlffmdgkjo
16:31:48.374 -> [WS] WebsocketsServer::recvHandshakeRequest: value = 13
16:31:48.374 -> [WS] WebsocketsServer::recvHandshakeRequest: value = gzip, deflate
16:31:48.374 -> [WS] WebsocketsServer::recvHandshakeRequest: value = de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
16:31:48.374 -> [WS] WebsocketsServer::recvHandshakeRequest: value = wKB6SmeYBrzZihXFi/Mhxg==
16:31:48.374 -> [WS] WebsocketsServer::recvHandshakeRequest: value = permessage-deflate; client_max_window_bits

 

 

 

 

Her is my Arduino Output.

 

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\admin\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Nextcloud\Eisenbahn_Fremo\arduino_Work\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=2341_003D -ide-version=10813 -build-path C:\Users\admin\AppData\Local\Temp\arduino_build_157696 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -verbose C:\Nextcloud\Eisenbahn_Fremo\Test2\_testkhoih\testeth\testeth.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\admin\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Nextcloud\Eisenbahn_Fremo\arduino_Work\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=2341_003D -ide-version=10813 -build-path C:\Users\admin\AppData\Local\Temp\arduino_build_157696 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\admin\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -verbose C:\Nextcloud\Eisenbahn_Fremo\Test2\_testkhoih\testeth\testeth.ino
Using board 'arduino_due_x_dbg' from platform in folder: C:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Using core 'arduino' from platform in folder: C:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Detecting libraries used...
"C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10813 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACT [... cut ...]
Using library SPI at version 1.0 in folder: C:\Users\admin\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\libraries\SPI
"C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-size" -A "C:\\Users\\admin\\AppData\\Local\\Temp\\arduino_build_157696/testeth.ino.elf"
Sketch uses 116108 bytes (22%) of program storage space. Maximum is 524288 bytes.
Atmel SMART device 0x285e0a60 found
Erase flash
done in 0.036 seconds

Write 119124 bytes to flash (466 pages)
 

 

 

 

 

 

 

Gesendet: Mittwoch, 05. August 2020 um 08:17 Uhr

jakespeed1311, 2020-08-05:
I'm glad it's going well, but unfortunately I can't make it work. I feel a little stupid.

 

 

Gesendet: Mittwoch, 05. August 2020 um 17:04 Uhr

jakespeed1311, 2020-08-05:
HI,

i removed the websoc code 

    
/*   WebsocketsClient client = SocketsServer.accept();
  
    if (client.available())
    {
      WebsocketsMessage msg = client.readBlocking();
  
      // log
      Serial.print("Got Message: ");
      Serial.println(msg.data());
  
      // return echo
      client.send("Echo: " + msg.data());
  
      // close the connection
      client.close();
    }*/

 

and the webserver runs. When i add the code and enable the serial output  s /W  then the loop ends with one W after that it ends.

But the websock lock like to run, or never coms back.

 

i add an serial print in look to see whats happen.

 

Serial.println(runEthernetServer);

 

And after this   WebsocketsClient client = SocketsServer.accept();

     Serial.println("Ww");    was not printed out.

also the webservwer was never answering.

 

I think that is the point to go in the websockets lib, i am right ?

 

 

 

due output

19:44:11.838 -> 
19:44:11.838 -> Starting WebSockets2_Generic SAMDUE-Ethernet_ServerAllFunctionsDemo on NMCSAM DUE
19:44:11.838 -> Ethernet using W5x00 and Ethernet Library
19:44:11.838 -> [SETUP] BOOT WAIT 4
19:44:12.821 -> [SETUP] BOOT WAIT 3
19:44:13.839 -> [SETUP] BOOT WAIT 2
19:44:14.822 -> [SETUP] BOOT WAIT 1
19:44:16.415 -> 
19:44:16.415 -> W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 10
19:44:16.415 -> W5100::init: W5500, SSIZE =4096
19:44:16.415 -> WebSockets Server IP address: 192.168.2.95
19:44:16.415 -> WebSockets Server Running and Ready on SAM DUE
19:44:16.415 -> IP address: 192.168.2.95, Port: 8080
19:44:16.415 -> 1
[... 419 lines cut ...]

#endif      //defines_h

 

 

 

 

 

 

 

 

 

 
 

Gesendet: Mittwoch, 05. August 2020 um 17:04 Uhr

khoih-prog, 2020-08-05:
You have to run the posted sample code (with the IP address modified) **successfully** to be sure your board / settings are OK before changing the code gradually to fix your purpose. 

**If not OK, you have to fix your hardware / settings first.** 

Do you have to use 

```
IPAddress serverIP(192, 168, 33, 85);
```

or it's better to use DHCP IP

```
 // start the ethernet connection and the server:
  // Use Static IP
  //Ethernet.begin(mac, serverIP);
  //Configure IP address via DHCP
  Ethernet.begin(mac);
```

**I can't help any further**

jakespeed1311, 2020-08-05:
ok I should probably try out more before I write
It seems to me that the websock only continues when it has a client.
But the web page is still not displayed.
When I connect and disconnect the websock the web page will build up. So somewhere in the websock there is a loop that does not release the client while it is there. That means the websock must release after time x. Or did I do something wrong with the ethernet library ?

Translated with www.DeepL.com/Translator (free version)

 

Thank you for helping.  I don't want to spam you with mails but i is my slow progres to understand whats going on.

 

 

 

 
 

Gesendet: Mittwoch, 05. August 2020 um 17:04 Uhr

jakespeed1311, 2020-08-05:
So that's the code and I have adjusted my IP that's not it and the hardware can't be it.

One thing must be clear the browser that opens the page will also be the receiver of the websocket. I want to display the logging data in a browse window. Probably it works because your clients are constantly querying the websocket and can do that ?

 

 
 

Gesendet: Mittwoch, 05. August 2020 um 20:10 Uhr

khoih-prog, 2020-08-05:
> Probably it works because your clients are constantly querying the websocket and can do that ?

If you specify your problem early, it'll be helpful and save time

OK,  if I understand correctly now, your problem is that 

1. the WebSocketServer is in blocking mode
2. you don't have any WebSockets Client
3. WS Server is blocked, waiting for Client before moving forward
3. The HTTP WebServer can't continue.

If so, I think I can fix the problem by modifying the [**WebSockets2_Generic Library**](https://github.com/khoih-prog/WebSockets2_Generic).

Please wait until I find out the fix and post the new release.

I'm moving the issue to the correct place of [**WebSockets2_Generic Library**](https://github.com/khoih-prog/WebSockets2_Generic) as  [**Running Http and Websocket Servers concurrently**](https://github.com/khoih-prog/WebSockets2_Generic/issues/1)

From now on, please post to the new place.

khoih-prog, 2020-08-06:
Hi @jakespeed1311 

The new Version v1.0.6 has been released with these new features to fix the blocking action of WebSockets Server that prevents the **concurrent running of WebServer and WebSockets Server you experienced so far.**

### New in [WebSockets2_Generic v1.0.6](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.0.6)

1. Add ***non-blocking WebSockets Server*** feature to enable WS Server and WebServer running ***concurently***. See [***Support as Http server and the Websockets server on Arduino DUE***](https://github.com/khoih-prog/WebSockets_Generic/issues/1) and [***Running Http and Websocket Servers concurrently***](https://github.com/khoih-prog/WebSockets2_Generic/issues/1). Thanks to bug report and persistence of [Jake](https://github.com/jakespeed1311).
2. Add non-blocking WebSockets Server and WebServer examples. 
3. Add Ethernet Library Patches
4. Add Arduino SAMD Packages_Patches to fix compiler errors when using [Standard Template Library STL](https://en.wikipedia.org/wiki/Standard_Template_Library)

Please try the example [SAMDUE-Ethernet_AdvancedWebServer](https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/Generic/Ethernet/SAMDUE/SAMDUE-Ethernet_AdvancedWebServer) to have feelings. Then modify it to fix your purpose.

Please be noted that your bug report and persistence, resulting in the new Version v1.0.6, has been noted in [Contributions-and-Thanks](https://github.com/khoih-prog/WebSockets2_Generic#contributions-and-thanks)