khoih-prog/WebSockets_Generic #2: No free space new client / Counting of clients wrong?

Algorab, 2020-06-04:
Try to setup a web socket server with wifinina.

When I try the example Generic_WebSocketServer_WiFiNINA.ino the
websocket connection is always closed. I try to connect with different clients. 
As far I can see the MKR WiFi 1010 should be supported since version 2.1.3.

It looks for me that the counting of the clients is not correct. I try to open only
one connection. See the log.

Wifi-version:
WiFiNINA firmware check.
Firmware version installed: 1.3.0
Latest firmware version available : 1.3.0
Check result: PASSED

websockets generic version: 2.2.2

[WS] 0 new client
[WS] 0 [handleHeader] RX: GET / HTTP/1.1
[WS] 1 new client
[WS] 0 [handleHeader] RX: Host: 10.X.X.X:81
[WS] 1 [handleHeader] RX: User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:77.0) Gecko/20100101 Firefox/77.0
[WS] 2 new client
[WS] 0 [handleHeader] RX: Accept: */*
[WS] 1 [handleHeader] RX: Accept-Language: de,en-US;q=0.7,en;q=0.3
[WS] 2 [handleHeader] RX: Accept-Encoding: gzip, deflate
[WS] 3 new client
[WS] 0 [handleHeader] RX: Sec-WebSocket-Version: 13
[WS] 1 [handleHeader] RX: Origin: moz-extension://79922fef-6785-6549-9f19-3efc822c7d29
[WS] 2 [handleHeader] RX: Sec-WebSocket-Protocol: arduino
[WS] 3 [handleHeader] RX: Sec-WebSocket-Extensions: permessage-deflate
[WS] 4 new client
[WS] 0 [handleHeader] RX: Sec-WebSocket-Key: 3yzKvswNEXx9sBqgHUK7qw==
[WS] 1 [handleHeader] RX: Connection: keep-alive, Upgrade
[WS] 2 [handleHeader] RX: Pragma: no-cache
[WS] 3 [handleHeader] RX: Cache-Control: no-cache
[WS] 4 [handleHeader] RX: Upgrade: websocket
[WS] [WS-Server] No free space new client
[WS] 0 client connection lost
[WS] 0 client disconnected.

khoih-prog, 2020-06-05:
Hi @Algorab 

I'll spend time to investigate the issue. Thanks for the useful info.

khoih-prog, 2020-06-05:
Hi @Algorab 

Currently, I'm a little busy and can't spend lot of time on the bug. 
I hope you can somehow help to speed up the process by doing this

1. Try the original [WebSocketServer for ESP8266](https://github.com/Links2004/arduinoWebSockets/tree/master/examples/esp8266/WebSocketServer) on ESP8266 or [WebSocketServer for ESP32](https://github.com/Links2004/arduinoWebSockets/tree/master/examples/esp32/WebSocketServer) on ESP32

2. If you experience the same problem, please post an issue on [arduinoWebSockets Issues](https://github.com/Links2004/arduinoWebSockets/issues)

As this library is currently new and just a port from ESP to other platforms, it inherits all the features and bugs from the original [arduinoWebSockets Library](https://github.com/Links2004/arduinoWebSockets).

Thanks,

Algorab, 2020-06-06:
Hi @khoih-prog,
switch to an ESP Arduino isn't a option for me due I have a special Omron development board which requires a MKR. I tried to investigate it and dig in the last days, but me C knowledge is rather thin, so I haven't found the cause yet.

khoih-prog, 2020-06-06:
Hi @Algorab 

Just spent some time to look at the issue. It seems **more complicated** than normal simple bug. It'll affect the WebSockets **Server** function, the WebSockets Client function is still OK.

The main issue is the original library WiFiNINA / WiFiNINA_Generic is not fully compatible with the original WebSockets library, originally written for ESP. It'll take much longer time to find out and solve the issue. Even it'll need a drastic measure to re-write / restructure either the WiFiNINA_Generic or WebSockets_Generic library.

So, don't expect this will be solved quickly. If you can find another quicker solution, I suggest you go with that.

The [Major issue with WiFiNINA - Arduino IoT 33](https://github.com/khoih-prog/WebSockets_Generic/issues/3) is another aspect of this same bug / issue.

Good Luck,

khoih-prog, 2020-07-16:
Hi @Algorab 

Please try the new [WebSockets2_Generic Library](https://github.com/khoih-prog/WebSockets2_Generic) I believe there is no WebSocket Server issue as this [WebSockets_Generic Library](https://github.com/khoih-prog/WebSockets_Generic).

This is terminal debug output when running [SAMDUE-Ethernet-Server](examples/Generic/Ethernet/SAMDUE/SAMDUE-Ethernet-Server) on SAM DUE board with Ethernet W5100 shield using EthernetLarge library, and receiving WebSockets connection from an ESP8266 board.

```
Starting WebSockets2_Generic SAMDUE-Ethernet-Server on SAM DUE
[SETUP] BOOT WAIT 4
[SETUP] BOOT WAIT 3
[SETUP] BOOT WAIT 2
[SETUP] BOOT WAIT 1

W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 10
W5100::init: W5100, SSIZE =4096
WebSockets Server Running and Ready on SAM DUE
IP address: 192.168.2.153, Port: 8080
H[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.153
[WS] WebsocketsServer::recvHandshakeRequest: value = ********
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from ESP8266
[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.153
[WS] WebsocketsServer::recvHandshakeRequest: value = ********
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from ESP8266
HHHHHHHHH
```

### Why do we need this [WebSockets2_Generic library](https://github.com/khoih-prog/WebSockets2_Generic)

Many Web services require WebSockets library, which is so far written only for ESP8266/ESP32 boards. The ESP boards so far rely on the [Markus Sattler's WebSockets Library](https://github.com/Links2004/arduinoWebSockets) to connect to Alexa via Sinric or SinricPro skills.

This [WebSockets2_Generic library](https://github.com/khoih-prog/WebSockets2_Generic) is based on and modified from [Gil Maimon's ArduinoWebSockets Library](https://github.com/gilmaimon/ArduinoWebsockets) to provide support to many more boards, such as ***Arduino SAMD21, Adafruit SAMD21/SAMD51, nRF52, STM32, Teensy, SAM DUE, etc.*** and enable those boards to use WebSockets services, including voice-control Alexa along with Blynk. Those supported boards can also run ***WebSockets Server.*** The WebSockets can be used with ***ESP’s WiFi, WiFiNINA, W5x00/ENC28J60/LAN8742A Ethernet, ESP8266/ESP32-AT modules/shields.***

---

#### New in v1.0.1

1. Add support to ***SAM DUE*** boards.
2. Add support to ***Ethernet W5x00, using either Ethernet, EthernetLarge or Ethernet2 library.***
3. Restructure code to provide flexibility to support many more WiFi/Ethernet modules/shields in the future. Please delete the *.cpp files, replaced by *.hpp files, in the src directory, if *.cpp files still exist after installing new version.

#### New in v1.0.0

1. Add support to ***nRF52*** boards, such as ***AdaFruit Feather nRF52832, nRF52840 Express, BlueFruit Sense, Itsy-Bitsy nRF52840 Express, Metro nRF52840 Express, NINA_B302_ublox, NINA_B112_ublox, etc.***.
2. Add support to ***SAMD51 (Itsy-Bitsy M4, Metro M4, Grand Central M4, Feather M4 Express, etc.)***.
3. Add support to ***SAMD21 (ZERO, MKR, NANO_33_IOT, M0, M0 Pro, AdaFruit CIRCUITPLAYGROUND_EXPRESS, etc.)***.
4. Add Alexa / SINRIC examples and supports to those boards as well as ESP8266/ESP32

khoih-prog, 2020-08-06:
The new [WebSockets2_Generic v1.0.6](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.0.6) has been released with these new features to fix the blocking action of WebSockets Server that prevents the **concurrent running of WebServer and WebSockets Server you experienced so far.**

### New in [WebSockets2_Generic v1.0.6](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.0.6)

1. Add ***non-blocking WebSockets Server*** feature to enable WS Server and WebServer running ***concurently***. See [***Support as Http server and the Websockets server on Arduino DUE***](https://github.com/khoih-prog/WebSockets_Generic/issues/1) and [***Running Http and Websocket Servers concurrently***](https://github.com/khoih-prog/WebSockets2_Generic/issues/1). Thanks to bug report and persistence of [Jake](https://github.com/jakespeed1311).
2. Add non-blocking WebSockets Server and WebServer examples. 
3. Add Ethernet Library Patches
4. Add Arduino SAMD Packages_Patches to fix compiler errors when using [Standard Template Library STL](https://en.wikipedia.org/wiki/Standard_Template_Library)

Please try the example [SAMD-AdvancedWebServer](https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/Generic/WiFiNINA/SAMD/SAMD-AdvancedWebServer) on Nano-33-IoT to have feelings. Then modify it to fix your purpose.

closed 2020-08-06 by khoih-prog