khoih-prog/WebSockets_Generic #3: Major issue with WiFiNINA - Arduino IoT 33

giliweb, 2020-06-05:
Try to run the  example Generic_WebSocketServer_WiFiNINA on my Arduino IoT 33.
It connects to my Home wifi, then it just starts spamming the DISCONNECTED event:
```
08:53:46.281 -> Start Generic_WebSocketServer_WiFiNINA
08:53:46.281 -> Used/default SPI pinout:
08:53:46.281 -> MOSI:11
08:53:46.281 -> MISO:12
08:53:46.281 -> SCK:13
08:53:46.281 -> SS:10
08:53:46.281 -> [SETUP] BOOT WAIT 4
08:53:47.300 -> [SETUP] BOOT WAIT 3
08:53:48.283 -> [SETUP] BOOT WAIT 2
08:53:49.279 -> [SETUP] BOOT WAIT 1
08:53:51.055 -> Please upgrade the firmware
08:53:51.055 -> Attempting to connect to SSID: Home
08:53:55.723 -> WiFi-begin: return1 = 3
08:53:55.723 -> WiFi-begin: return2 = 3
08:53:55.723 -> SSID: Home
08:53:55.723 -> WebSockets Server IP address: 192.168.68.110
08:53:55.723 -> signal strength (RSSI):-77 dBm
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
08:53:55.723 -> [0] Disconnected!
```
It continues like that until arduino ide just crashes. 
I'm not even trying to connect another device at this point, cos i can't.
Can you please help with this issue? I found this library has a lot of potential but i can't use it. Or maybe i can contribute if you point me in the right direction, thanks.

giliweb, 2020-06-05:
This is what i get if i enable DEBUG 4
```
09:06:26.795 -> [WS] 0 new client
09:06:26.795 -> [WS] 0 client connection lost
09:06:26.795 -> [WS] 0 client disconnected.
09:06:26.795 -> [0] Disconnected!
09:06:26.795 -> [WS] 0 new client
09:06:26.795 -> [WS] 0 client connection lost
09:06:26.795 -> [WS] 0 client disconnected.
09:06:26.795 -> [0] Disconnected!
09:06:26.795 -> [WS] 0 new client
09:06:26.795 -> [WS] 0 client connection lost
09:06:26.795 -> [WS] 0 client disconnected.
09:06:26.795 -> [0] Disconnected!
09:06:26.795 -> [WS] 0 new client
09:06:26.795 -> [WS] 0 client connection lost
09:06:26.795 -> [WS] 0 client disconnected.
09:06:26.795 -> [0] Disconnected!
09:06:26.795 -> [WS] 0 new client
09:06:26.795 -> [WS] 0 client connection lost
09:06:26.795 -> [WS] 0 client disconnected.
09:06:26.795 -> [0] Disconnected!

```

khoih-prog, 2020-06-05:
Hi @giliweb 

The messages are just for debugging, and in the current example, the debug level is set to the highest level. It's possible that **too many unnecessary messages** sent to Serial port creates the **unresponsiveness and crash.**

Can you try by reduce the debug level to minimum (at the beginning of sketch), then inform if it's OK

```
#define _WEBSOCKETS_LOGLEVEL_     0
```

I'll spend time to go deeper to find the bug and fix.

giliweb, 2020-06-05:
Hi @khoih-prog thanks for your quick reply.
My main concern is not the crash of the serial monitor.
The example is continuously triggering the connected/disconnected event, event though no clients are still connected. That's the issue.

khoih-prog, 2020-06-05:
@giliweb 

Understand that, that's why I said 

> I'll spend time to go deeper to find the bug and fix.

khoih-prog, 2020-06-05:
Hi @giliweb 

Currently, I'm a little busy and can't spend lot of time on the bug. 
I hope you can somehow help to speed up the process by doing this

1. Try the original [WebSocketServer for ESP8266](https://github.com/Links2004/arduinoWebSockets/tree/master/examples/esp8266/WebSocketServer) on ESP8266 or [WebSocketServer for ESP32](https://github.com/Links2004/arduinoWebSockets/tree/master/examples/esp32/WebSocketServer) on ESP32

2. If you experience the same problem, please post an issue on [arduinoWebSockets Issues](https://github.com/Links2004/arduinoWebSockets/issues)

As this library is currently new and just a port from ESP to other platforms, it inherits all the features and bugs from the original [arduinoWebSockets Library](https://github.com/Links2004/arduinoWebSockets).

Thanks,

giliweb, 2020-06-05:
I should have an esp32 board somewhere, I'll give it a try

khoih-prog, 2020-06-06:
Hi @giliweb 

Just spent some time to look at the issue. It seems **more complicated** than normal simple bug. It'll affect the WebSockets **Server** function, the WebSockets Client function is still OK.

The main issue is the original library WiFiNINA / WiFiNINA_Generic is **not fully compatible** with the original WebSockets library, originally written for ESP. It'll take much longer time to find out and solve the issue. Even it'll need a drastic measure to re-write / restructure either the WiFiNINA_Generic or WebSockets_Generic library.

So, don't expect this will be solved quickly. If you can find another quicker solution, I suggest you go with that.

The [No free space new client / Counting of clients wrong?](https://github.com/khoih-prog/WebSockets_Generic/issues/2) is another aspect of this same bug / issue.

Good Luck,

khoih-prog, 2020-07-16:
Hi @giliweb 

Please try the new [WebSockets2_Generic Library](https://github.com/khoih-prog/WebSockets2_Generic) I believe there is no WebSocket Server issue as this [WebSockets_Generic Library](https://github.com/khoih-prog/WebSockets_Generic).

This is terminal debug output when running [SAMDUE-Ethernet-Server](examples/Generic/Ethernet/SAMDUE/SAMDUE-Ethernet-Server) on SAM DUE board with Ethernet W5100 shield using EthernetLarge library, and receiving WebSockets connection from an ESP8266 board.

```
Starting WebSockets2_Generic SAMDUE-Ethernet-Server on SAM DUE
[SETUP] BOOT WAIT 4
[SETUP] BOOT WAIT 3
[SETUP] BOOT WAIT 2
[SETUP] BOOT WAIT 1

W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 10
W5100::init: W5100, SSIZE =4096
WebSockets Server Running and Ready on SAM DUE
IP address: 192.168.2.153, Port: 8080
H[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.153
[WS] WebsocketsServer::recvHandshakeRequest: value = ********
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from ESP8266
[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.153
[WS] WebsocketsServer::recvHandshakeRequest: value = ********
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from ESP8266
HHHHHHHHH
```

### Why do we need this [WebSockets2_Generic library](https://github.com/khoih-prog/WebSockets2_Generic)

Many Web services require WebSockets library, which is so far written only for ESP8266/ESP32 boards. The ESP boards so far rely on the [Markus Sattler's WebSockets Library](https://github.com/Links2004/arduinoWebSockets) to connect to Alexa via Sinric or SinricPro skills.

This [WebSockets2_Generic library](https://github.com/khoih-prog/WebSockets2_Generic) is based on and modified from [Gil Maimon's ArduinoWebSockets Library](https://github.com/gilmaimon/ArduinoWebsockets) to provide support to many more boards, such as ***Arduino SAMD21, Adafruit SAMD21/SAMD51, nRF52, STM32, Teensy, SAM DUE, etc.*** and enable those boards to use WebSockets services, including voice-control Alexa along with Blynk. Those supported boards can also run ***WebSockets Server.*** The WebSockets can be used with ***ESP’s WiFi, WiFiNINA, W5x00/ENC28J60/LAN8742A Ethernet, ESP8266/ESP32-AT modules/shields.***

---

#### New in v1.0.1

1. Add support to ***SAM DUE*** boards.
2. Add support to ***Ethernet W5x00, using either Ethernet, EthernetLarge or Ethernet2 library.***
3. Restructure code to provide flexibility to support many more WiFi/Ethernet modules/shields in the future. Please delete the *.cpp files, replaced by *.hpp files, in the src directory, if *.cpp files still exist after installing new version.

#### New in v1.0.0

1. Add support to ***nRF52*** boards, such as ***AdaFruit Feather nRF52832, nRF52840 Express, BlueFruit Sense, Itsy-Bitsy nRF52840 Express, Metro nRF52840 Express, NINA_B302_ublox, NINA_B112_ublox, etc.***.
2. Add support to ***SAMD51 (Itsy-Bitsy M4, Metro M4, Grand Central M4, Feather M4 Express, etc.)***.
3. Add support to ***SAMD21 (ZERO, MKR, NANO_33_IOT, M0, M0 Pro, AdaFruit CIRCUITPLAYGROUND_EXPRESS, etc.)***.
4. Add Alexa / SINRIC examples and supports to those boards as well as ESP8266/ESP32

khoih-prog, 2020-08-06:
The new [WebSockets2_Generic v1.0.6](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.0.6) has been released with these new features to fix the blocking action of WebSockets Server that prevents the **concurrent running of WebServer and WebSockets Server you experienced so far.**

### New in [WebSockets2_Generic v1.0.6](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.0.6)

1. Add ***non-blocking WebSockets Server*** feature to enable WS Server and WebServer running ***concurently***. See [***Support as Http server and the Websockets server on Arduino DUE***](https://github.com/khoih-prog/WebSockets_Generic/issues/1) and [***Running Http and Websocket Servers concurrently***](https://github.com/khoih-prog/WebSockets2_Generic/issues/1). Thanks to bug report and persistence of [Jake](https://github.com/jakespeed1311).
2. Add non-blocking WebSockets Server and WebServer examples. 
3. Add Ethernet Library Patches
4. Add Arduino SAMD Packages_Patches to fix compiler errors when using [Standard Template Library STL](https://en.wikipedia.org/wiki/Standard_Template_Library)

Please try the example [SAMD-AdvancedWebServer](https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/Generic/WiFiNINA/SAMD/SAMD-AdvancedWebServer) on Nano-33-IoT to have feelings. Then modify it to fix your purpose.

closed 2020-08-06 by khoih-prog