khoih-prog/WebSockets_Generic #4: Generic_WebSocketServer_LEDcontrol_W5500 run error

lixy123, 2020-06-20:
/examples/Generic/W5500/Generic_WebSocketServer_LEDcontrol_W5500
run error:
WebSocketBadStatusException: Handshake status 400 Bad Request
websocket._exceptions.WebSocketBadStatusException: Handshake status 400 Bad Request

Hardware : teensy3.6,ws5500,
arduino version:  1.8.12 
browser:  google chrome  83.0.4103.106 64.

khoih-prog, 2020-06-20:
Hi @lixy123 

Currently, the WebSocketServer features of this library is not working due to similar issues in

1. [No free space new client / Counting of clients wrong? #2](https://github.com/khoih-prog/WebSockets_Generic/issues/2)
2. [Major issue with WiFiNINA - Arduino IoT 33 #3](https://github.com/khoih-prog/WebSockets_Generic/issues/3)

The timeline when to fix is not determined yet.

I'll post a warning in README about the WebSocketServer issues.

khoih-prog, 2020-07-16:
Hi @lixy123 

Please try the new [WebSockets2_Generic Library](https://github.com/khoih-prog/WebSockets2_Generic) I believe there is no WebSocket Server issue as this [WebSockets_Generic Library](https://github.com/khoih-prog/WebSockets_Generic).

This is terminal debug output when running [SAMDUE-Ethernet-Server](examples/Generic/Ethernet/SAMDUE/SAMDUE-Ethernet-Server) on SAM DUE board with Ethernet W5100 shield using EthernetLarge library, and receiving WebSockets connection from an ESP8266 board.

```
Starting WebSockets2_Generic SAMDUE-Ethernet-Server on SAM DUE
[SETUP] BOOT WAIT 4
[SETUP] BOOT WAIT 3
[SETUP] BOOT WAIT 2
[SETUP] BOOT WAIT 1

W5100 init, using SS_PIN_DEFAULT = 10, new ss_pin = 10, W5100Class::ss_pin = 10
W5100::init: W5100, SSIZE =4096
WebSockets Server Running and Ready on SAM DUE
IP address: 192.168.2.153, Port: 8080
H[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.153
[WS] WebsocketsServer::recvHandshakeRequest: value = ********
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from ESP8266
[WS] WebsocketsServer::recvHandshakeRequest: value = 192.168.2.153
[WS] WebsocketsServer::recvHandshakeRequest: value = ********
[WS] WebsocketsServer::recvHandshakeRequest: value = websocket
[WS] WebsocketsServer::recvHandshakeRequest: value = Upgrade
[WS] WebsocketsServer::recvHandshakeRequest: value = 13
[WS] WebsocketsServer::recvHandshakeRequest: value = TinyWebsockets Client
[WS] WebsocketsServer::recvHandshakeRequest: value = Basic 
[WS] WebsocketsServer::recvHandshakeRequest: value = https://github.com/khoih-prog/Websockets2_Generic
Got Message: Hello to Server from ESP8266
HHHHHHHHH
```

### Why do we need this [WebSockets2_Generic library](https://github.com/khoih-prog/WebSockets2_Generic)

Many Web services require WebSockets library, which is so far written only for ESP8266/ESP32 boards. The ESP boards so far rely on the [Markus Sattler's WebSockets Library](https://github.com/Links2004/arduinoWebSockets) to connect to Alexa via Sinric or SinricPro skills.

This [WebSockets2_Generic library](https://github.com/khoih-prog/WebSockets2_Generic) is based on and modified from [Gil Maimon's ArduinoWebSockets Library](https://github.com/gilmaimon/ArduinoWebsockets) to provide support to many more boards, such as ***Arduino SAMD21, Adafruit SAMD21/SAMD51, nRF52, STM32, Teensy, SAM DUE, etc.*** and enable those boards to use WebSockets services, including voice-control Alexa along with Blynk. Those supported boards can also run ***WebSockets Server.*** The WebSockets can be used with ***ESP’s WiFi, WiFiNINA, W5x00/ENC28J60/LAN8742A Ethernet, ESP8266/ESP32-AT modules/shields.***

---

#### New in v1.0.1

1. Add support to ***SAM DUE*** boards.
2. Add support to ***Ethernet W5x00, using either Ethernet, EthernetLarge or Ethernet2 library.***
3. Restructure code to provide flexibility to support many more WiFi/Ethernet modules/shields in the future. Please delete the *.cpp files, replaced by *.hpp files, in the src directory, if *.cpp files still exist after installing new version.

#### New in v1.0.0

1. Add support to ***nRF52*** boards, such as ***AdaFruit Feather nRF52832, nRF52840 Express, BlueFruit Sense, Itsy-Bitsy nRF52840 Express, Metro nRF52840 Express, NINA_B302_ublox, NINA_B112_ublox, etc.***.
2. Add support to ***SAMD51 (Itsy-Bitsy M4, Metro M4, Grand Central M4, Feather M4 Express, etc.)***.
3. Add support to ***SAMD21 (ZERO, MKR, NANO_33_IOT, M0, M0 Pro, AdaFruit CIRCUITPLAYGROUND_EXPRESS, etc.)***.
4. Add Alexa / SINRIC examples and supports to those boards as well as ESP8266/ESP32

khoih-prog, 2020-08-06:
The new [WebSockets2_Generic v1.0.6](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.0.6) has been released with these new features to fix the blocking action of WebSockets Server that prevents the **concurrent running of WebServer and WebSockets Server you experienced so far.**

### New in [WebSockets2_Generic v1.0.6](https://github.com/khoih-prog/WebSockets2_Generic/releases/tag/v1.0.6)

1. Add ***non-blocking WebSockets Server*** feature to enable WS Server and WebServer running ***concurently***. See [***Support as Http server and the Websockets server on Arduino DUE***](https://github.com/khoih-prog/WebSockets_Generic/issues/1) and [***Running Http and Websocket Servers concurrently***](https://github.com/khoih-prog/WebSockets2_Generic/issues/1). Thanks to bug report and persistence of [Jake](https://github.com/jakespeed1311).
2. Add non-blocking WebSockets Server and WebServer examples. 
3. Add Ethernet Library Patches
4. Add Arduino SAMD Packages_Patches to fix compiler errors when using [Standard Template Library STL](https://en.wikipedia.org/wiki/Standard_Template_Library)

Please try the example [SAMDUE-Ethernet_AdvancedWebServer](https://github.com/khoih-prog/WebSockets2_Generic/tree/master/examples/Generic/Ethernet/SAMDUE/SAMDUE-Ethernet_AdvancedWebServer) or similar to have feelings. Then modify it to fix your purpose.

closed 2020-08-06 by khoih-prog