khoih-prog/WebSockets_Generic #9: [Question]

qlalfdu, 2021-05-07:
I need to use ESP32 as a websocket over ethernet without wifi.
The environment cannot use wifi.
I need a websocket connection from my laptop to ESP32 via ethernet.
How can this be possible?
Even if I try to change the example, the server is connected and the web page is loaded, but the websocket cannot be connected.

Thank you in advance.
[image]

khoih-prog, 2021-05-07:
This [**WebSockets** library](https://github.com/khoih-prog/WebSockets_Generic) only works for ESP826/EP32 using native WiiFi. Check  [**Important Notes**](https://github.com/khoih-prog/WebSockets_Generic#important-notes). I suggest you try the [**WebSockets2_Generic Library**](https://github.com/khoih-prog/WebSockets2_Generic).

---

#### Important Notes

1. Currently, the WebSocketServer feature is usable only for ESP8266/ESP32  using native WiFi.
2. The timeline when to fix and reintroduce the WebSocketServer feature to other boards is not determined yet.
3. Please use the new [**WebSockets2_Generic Library**](https://github.com/khoih-prog/WebSockets2_Generic) if WebSocketServer is necessary. See [Issue 2](https://github.com/khoih-prog/WebSockets_Generic/issues/2), [Issue 3](https://github.com/khoih-prog/WebSockets_Generic/issues/3) and [Issue 4](https://github.com/khoih-prog/WebSockets_Generic/issues/4)

closed 2021-05-07 by khoih-prog

Shivdoke39, 2022-11-30:
client.onMessage([&](WebsocketsMessage message)
  {
   if 
   (strcmp(message.data(), "flashlight") == 0)
   {
    fl_state = fl_state ? 0 : 1;
    Serial.printf("Toggling fashlight to %u\n", fl_state);
    digitalWrite(fl, fl_state);
   }else{
    Serial.print("Got Message: ");
    Serial.println(message.data());
  });
errors:::::
//cannot convert 'websockets2_generic::WSInterfaceString' {aka 'String'} to 'const char*'