khoih-prog/WebSockets_Generic #10: WiFi101 and Arduino MKR1000.

Josua2012, 2021-05-22:
Please add support for WiFi101 and Arduino MKR1000. 👍 
Thank you. 
Regards,
Jose

khoih-prog, 2021-05-22:
Please be ready to test the new release to support WiFi101 and MKR1000 / MKRWIFI1010.

I'll be releasing it within 1/2 hr.

khoih-prog, 2021-05-22:
You have to use the new release with latest [**Forked WiFi101**](https://github.com/khoih-prog/WiFi101) to suppress many compiler's warnings relating to `"deleting object of polymorphic class type"`

Please download and test the [**Forked WiFi101**](https://github.com/khoih-prog/WiFi101) first.

Josua2012, 2021-05-22:
You are amazing. Truly amazing.
Many thanks.
Now it is very late here in Spain, but tomorrow, first thing in the morning, I will do all the checks I can.
Again, thank you so much.
You are a crack. 💯 🥇

Josua2012, 2021-05-22:
I could not resist trying the new version of WiFi101 and it seems that everything works without any errors. 💯

khoih-prog, 2021-05-22:
Great. Almost done. Just modifying READ now. Within 5 mins. Are you staying??

Josua2012, 2021-05-22:
Sure

Josua2012, 2021-05-22:
Khoi,
Will you put some example of socket.io, websocket client, with WiFi101?
I have also observed that in the Websocket2 examples it automatically detected the type of board, however in Websocket that does not happen. Is it for a certain reason? 
Thanks, thanks ....

khoih-prog, 2021-05-22:
Ready now. Also check [**Contributions and Thanks**](https://github.com/khoih-prog/WebSockets_Generic#contributions-and-thanks)

The Socket.IO example s here

[WebSocketClientSocketIO_WiFi101](https://github.com/khoih-prog/WebSockets_Generic/tree/master/examples/WiFi101/WebSocketClientSocketIO_WiFi101)

---

### Major Release v2.5.0

1. Add support to SAMD_MKR1000 and SAMD_MKRWIFI1010 using WiFi101.  To be used with [**Forked WiFi101** library](https://github.com/khoih-prog/WiFi101)
2. Clean compiler's warnings
3. Add examples for WiFi101

khoih-prog, 2021-05-22:
All the WiFi101 examples are here [**For WiFi101**](https://github.com/khoih-prog/WebSockets_Generic#for-wifi101)

khoih-prog, 2021-05-22:
Waiting for your preliminary test results now !!!

Josua2012, 2021-05-22:
You didn't need to name me, if you've done it all yourself.
You are a fabulous programming machine. 💯

khoih-prog, 2021-05-22:
Don't say so. Without your request and test, I could do nothing to add new  support. I always note to remember everybody's help, no matter what.

khoih-prog, 2021-05-22:
> I have also observed that in the Websocket2_Generic examples it automatically detected the type of board, however in Websocket_Generic that does not happen. Is it for a certain reason?

No good reason at all, just the [WebSockets2_Generic](https://github.com/khoih-prog/WebSockets2_Generic) is much younger than [WebSockets_Generic](https://github.com/khoih-prog/WebSockets_Generic). And I added more features to detect the board name for easier to debug (important for library maintainer/writer to know which board and core user selected and if it's correct or supported, etc.).

It's easier to add, just copy the `defines.h` from **Websocket2_Generic** examples to relating **Websocket_Generic** examples.
Might be next release, if I remember and have time. Not in rushing like this release. ;=))

Josua2012, 2021-05-22:
Do not worry. I was just curious if there was any important reason. 
Good to know that `defines.h` is compatible with both libraries.

Josua2012, 2021-05-22:
I also want to test your WiFiManager library, I think it might be interesting to me.
Do you know if it is compatible with WiFi101 and Arduino MKR1000?

khoih-prog, 2021-05-22:
> I also want to test your WiFiManager library, I think it might be interesting to me.
> Do you know if it is compatible with WiFi101 and Arduino MKR1000?

It's ready here : [**WiFiManager_Generic_Lite** library](https://github.com/khoih-prog/WiFiManager_Generic_Lite). 
Test to see if there is any issue.

You also have to install and use my [**FlashStorage_SAMD** library](https://github.com/khoih-prog/FlashStorage_SAMD), which you can also use to store **persistent** data, necessary for you SAMD applications

khoih-prog, 2021-05-22:
The examples written specifically for **WiFi101** and **MKR1000** / MKR1010WiFi

1. [**MKR1000_WiFi101**](https://github.com/khoih-prog/WiFiManager_Generic_Lite/tree/main/examples/MKR1000_WiFi101)
2. [**MKR1000_WiFi101_MQTT**](https://github.com/khoih-prog/WiFiManager_Generic_Lite/tree/main/examples/MKR1000_WiFi101_MQTT)

khoih-prog, 2021-05-22:
Still waiting for your test results. Are you testing it now? If not, I have to move on to another work.

Josua2012, 2021-05-22:
First test.
The following warnings are still displayed:

[image]

It seems to be successfully connecting to the server, however the client is disconnecting, I have to investigate further to see if it is my problem. 

[image]

Josua2012, 2021-05-22:
By the way, thanks again for the fixes in the WiFiManager_Generic_Lite library.
As soon as I finish testing Websockets_Generic library I will start testing the WiFiManager library. :)

khoih-prog, 2021-05-22:
> The following warnings are still displayed:

Those are **intentional** and informational warnings, so that you know what you're doing and if the settings are correct. Just ignore them.

Josua2012, 2021-05-23:
To make the tests easier I have created a small socket.io server
I have encountered quite a few performance problems:

**Your library is supporting an old version of socket.io, version 3, currently it is version 4, and I had to enable its compatibility in the test server that I have done.**

[image]

**Now the server tells me that a client has connected, the problem is that even though there is only one client, the server is receiving multiple client connections.**

[image]

[image]

**Neither the sending of the messages to the server works. The messages do not reach him.** 

[image]

Can you do something with these kinds of errors? 

Thanks.

khoih-prog, 2021-05-23:
> Now the server tells me that a client has connected, the problem is that even though there is only one client, the server is receiving multiple client connections.
> 
> Neither the sending of the messages to the server works. The messages do not reach him.

Do you have another kind of board to test if this is happens only to MKR1000 ot WiFi101???

> Your library is supporting an old version of socket.io, version 3, currently it is version 4, and I had to enable its compatibility in the test server that I have done.

I'll have a look to update to Socket.IO v4

khoih-prog, 2021-05-23:
Can you copy the whole terminal output **text**, **not paste the image** for easier check.

Josua2012, 2021-05-23:
No. Unfortunately I don't have any other board.  :(

khoih-prog, 2021-05-23:
Can you test with other online Socket.IO server ?

Josua2012, 2021-05-23:
I could send you the little server.
Do you know nodejs and how to install the server?
You can easily install it locally. 
[websocket-server.zip](https://github.com/khoih-prog/WebSockets_Generic/files/6527353/websocket-server.zip)

Josua2012, 2021-05-23:
In windows:
1) Install nodejs
2) Go to the project folder that I have sent you in CMD
3) Run: npm i
4) Run: node app

Josua2012, 2021-05-23:
The server that I have sent you works perfectly, you can test it with the chrome browser by putting: localhost: 8080
And if you put any text in the input and send it with the "Enviar" button, you will see that the server receives it in the console.

khoih-prog, 2021-05-23:
The terminal output when running Nano-33-IoT with WiFiNINA. Not OK. Is that the same as your results?

```
Start WebSocketClientSocketIO_NINA on SAMD_NANO_33_IOT
WebSockets_Generic v2.5.0
Used/default SPI pinout:
MOSI:11
MISO:12
SCK:13
SS:10
Attempting to connect to SSID: HueNet1
[WS] [WS-Client] Connect ws...
[WS] [WS-Client] connectedCb
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=3&transport=polling HTTP/1.1
Host: 192.168.2.30:8080
Connection: keep-alive
Origin: file://
User-Agent: arduino-WebSocket-Client

[WS] [WS-Client] [sendHeader] Sending header... Done (us):2451
[WS] [WS-Client][handleHeader] RX:HTTP/1.1 200 OK
[WS] [WS-Client][handleHeader] RX:Content-Type: text/plain; charset=UTF-8
[WS] [WS-Client][handleHeader] RX:Content-Length: 100
[WS] [WS-Client][handleHeader] RX:Date: Sun, 23 May 2021 00:38:56 GMT
[WS] [WS-Client][handleHeader] RX:Connection: keep-alive
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=3
[WS] [WS-Client][handleHeader] - cKey:W
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:0
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:
[WS] [WS-Client][handleHeader] no Websocket connection close.
[WS] [WS-Client] client disconnected.
[IOc] Disconnected
[WS] [WS-Client] Connect ws...
[WS] [WS-Client] Connect ws...
[WS] [WS-Client] connectedCb
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=3&transport=polling HTTP/1.1
Host: 192.168.2.30:8080
Connection: keep-alive
Origin: file://
User-Agent: arduino-WebSocket-Client

[WS] [WS-Client] [sendHeader] Sending header... Done (us):2371
[WS] [WS-Client][handleHeader] RX:HTTP/1.1 200 OK
[WS] [WS-Client][handleHeader] RX:Content-Type: text/plain; charset=UTF-8
[WS] [WS-Client][handleHeader] RX:Content-Length: 100
[WS] [WS-Client][handleHeader] RX:Date: Sun, 23 May 2021 00:39:04 GMT
[WS] [WS-Client][handleHeader] RX:Connection: keep-alive
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=3
[WS] [WS-Client][handleHeader] - cKey:R
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:0
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:
[WS] [WS-Client][handleHeader] no Websocket connection close.
[WS] [WS-Client] client disconnected.
[IOc] Disconnected
[WS] [WS-Client] Connect ws...
[WS] [WS-Client] connectedCb
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=3&transport=polling HTTP/1.1
Host: 192.168.2.30:8080
Connection: keep-alive
Origin: file://
User-Agent: arduino-WebSocket-Client
```

I'll move the test discussion into a private repo as they are TLTR; and not necessary for many users.
Will invite you in.

khoih-prog, 2021-05-23:
Just sending invitation to the private repo for discussion. Accept and go there.
Seems lengthy.

khoih-prog, 2021-05-23:
HI,

Please test new [**WebSockets_Generic** Releases v2.5.1](https://github.com/khoih-prog/WebSockets_Generic/releases/tag/v2.5.1)

I used it to test your server and connections are OK.

### Boards:

#### 1. Nano-33-IoT with WiFiNINA

```
Start WebSocketClientSocketIO_NINA on SAMD_NANO_33_IOT
WebSockets_Generic v2.5.1
Used/default SPI pinout:
MOSI:11
MISO:12
SCK:13
SS:10
Attempting to connect to SSID: HueNet1
SSID: HueNet1
WebSockets Client IP Address: 192.168.2.114
signal strength (RSSI):-19 dBm
Connecting to WebSockets Server @ IP address: 192.168.2.30, port: 8080
[WS] WebSockets_Generic v2.5.1
[WS] [wsIOc] found EIO=4 disable EIO ping on client
[WS] [WS-Client] Connect ws...
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=polling HTTP/1.1
Host: 192.168.2.30:8080
Connection: keep-alive
Origin: file://
User-Agent: arduino-WebSocket-Client

[WS] [WS-Client] [sendHeader] Sending header... Done (us):2586
[WS] [WS-Client][handleHeader] RX:HTTP/1.1 200 OK
[WS] [WS-Client][handleHeader] RX:Content-Type: text/plain; charset=UTF-8
[WS] [WS-Client][handleHeader] RX:Content-Length: 97
[WS] [WS-Client][handleHeader] RX:Date: Sun, 23 May 2021 04:04:06 GMT
[WS] [WS-Client][handleHeader] RX:Connection: keep-alive
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=4
[WS] [WS-Client][handleHeader] - cKey:Y
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:1
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
[WS] [WS-Client] Connect ws...
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=polling HTTP/1.1
Host: 192.168.2.30:8080
Connection: keep-alive
Origin: file://
User-Agent: arduino-WebSocket-Client
```

---

#### 2.  NRF52840_FEATHER with W5x00

```
[... 63 lines cut ...]
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:0
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:
[WS] [wsIOc] Disconnected!
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=polling HTTP/1.1
Host: 192.168.2.30:8080
Connection: keep-alive
Origin: file://
User-Agent: arduino-WebSocket-Client
```

---

### Release v2.5.1

1. Default to **EIO4 for Socket.IO**.
2. Permit changing reconnectInterval in Socket.IO. Default is 0.5s. Set in Socket.IO examples to 10s.
2. Better debug settings

khoih-prog, 2021-05-23:
OK. I detect the problem of not sending msg, and will work to solve it. Be patient.

khoih-prog, 2021-05-24:
Please test the new [**WebSockets_Generic** Releases v2.6.0](https://github.com/khoih-prog/WebSockets_Generic/releases/tag/v2.6.0) on **MKR1000 + WiFi101**

---

### Major Release v2.6.0

1. Fix **breaking problem with SocketIO**
2. Add setExtraHeaders function to SocketIO
3. Clean possible compiler's warnings

khoih-prog, 2021-05-24:
Test results

### WebSocketClientSocketIO_NINA on SAMD_NANO_33_IOT

#### 1. Client

```
Start WebSocketClientSocketIO_NINA on SAMD_NANO_33_IOT
WebSockets_Generic v2.6.0
Used/default SPI pinout:
MOSI:11
MISO:12
SCK:13
SS:10
Attempting to connect to SSID: HueNet1
SSID: HueNet1
WebSockets Client IP Address: 192.168.2.114
signal strength (RSSI):-18 dBm
Connecting to WebSockets Server @ IP address: 192.168.2.30, port: 8080
[WS] WebSockets_Generic v2.6.0
[WS] [wsIOc] found EIO=4 disable EIO ping on client
[WS] [WS-Client] Connect ws...
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] sendHeader: client->cKey = 7nb1qwQBvz58aozmAyBmag==
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=polling HTTP/1.1
Host: 192.168.2.30:8080
Connection: keep-alive
Origin: file://
User-Agent: arduino-WebSocket-Client

[WS] [WS-Client] [sendHeader] Sending header... Done (us):2669
[WS] [WS-Client][handleHeader] RX:HTTP/1.1 200 OK
[WS] [WS-Client][handleHeader] RX:Content-Type: text/plain; charset=UTF-8
[WS] [WS-Client][handleHeader] RX:Content-Length: 97
[WS] [WS-Client][handleHeader] RX:Date: Mon, 24 May 2021 02:19:53 GMT
[WS] [WS-Client][handleHeader] RX:Connection: keep-alive
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=4
[WS] [WS-Client][handleHeader] - cKey:7nb1qwQBvz58aozmAyBmag==
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:1
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:
[WS] [WS-Client][handleHeader] Still missing cSessionId try Socket.IO
[WS] [WS-Client][handleHeader] socket.io json: 0{"sid":"WdrsvWrSeNmPDSlFAAAa","upgrades":["websocket"],"pingInterval":25000,"pingTimeout":20000}
[WS] [WS-Client][handleHeader] - cSessionId: WdrsvWrSeNmPDSlFAAAa
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=4
[WS] [WS-Client][handleHeader] - cKey:7nb1qwQBvz58aozmAyBmag==
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:1
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:WdrsvWrSeNmPDSlFAAAa
[WS] [WS-Client][handleHeader] found cSessionId
[WS] [WS-Client] [sendHeader] Sending header...
[WS] sendHeader: client->cKey = 6b1sIaeSY4sl6HQqSLZaKQ==
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=websocket&sid=WdrsvWrSeNmPDSlFAAAa HTTP/1.1
[... 157 lines cut ...]
[WS] [wsIOc] get pong
[WS] [wsIOc] connected: len = 30
[WS] [wsIOc] data: {"sid":"cSx3YyFrDOfxUW8hAAAd"}
["event_name",{"now":30001}]
[WS] [wsIOc] get ping send pong:3
[WS] [WS-Client] Sending HB ping
["event_name",{"now":60002}]
```

---

#### 2. Server

```
kh@kh-Inspiron-3593:~/Downloads/Nodejs/New_SIO$ nodejs app
Server launched on port 8080
Connected
cSx3YyFrDOfxUW8hAAAd
JWT token test:  { host: '192.168.2.30:8080',
  connection: 'keep-alive',
  authorization: '1234567890',
  'user-agent': 'arduino-WebSocket-Client' }
Message from Client :  { now: 30001 }
Message from Client :  { now: 60002 }
```

closed 2021-05-24 by khoih-prog