khoih-prog/WebSockets_Generic #12: Socket.io client on Teensy Native Ethernet not working on Websockets_Generic

mechatroniks-git, 2021-06-22:
Describe the bug
Getting compiler warnings, not seeing any packets from Teensy using this code, only modified the IP server address(192.168.1.187) and Port (3000)

Steps to Reproduce
Steps to reproduce the behavior. Including the MRE sketches

Expected behavior
Expected to see TCP packets from teensy on Wireshark

Actual behavior
Can ping teensy, and see the pings on wireshark

Debug and AT-command log (if applicable)
Arduino compiler

Teensy_WebSocketClientSocketIO_NativeEthernet:84: warning: #warning Using NativeEthernet lib for Teensy 4.1. Must also use Teensy Packages Patch or error 
   #warning Using NativeEthernet lib for Teensy 4.1. Must also use Teensy Packages Patch or error
    ^
In file included from C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic.h:56:0,
                 from C:\Users\me\GitHub\Arduino\Teensy_WebSocketClientSocketIO_NativeEthernet\Teensy_WebSocketClientSocketIO_NativeEthernet.ino:126:
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSockets_Generic.h:231:4: warning: #warning Use Teensy in WebSockets_Generic [-Wcpp]
   #warning Use Teensy in WebSockets_Generic
    ^
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSockets_Generic.h:516:4: warning: #warning Using Teensy 4.1 NativeEthernet Library [-Wcpp]
   #warning Using Teensy 4.1 NativeEthernet Library
    ^
In file included from C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSockets_Generic.h:718:0,
                 from C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic.h:56,
                 from C:\Users\me\GitHub\Arduino\Teensy_WebSocketClientSocketIO_NativeEthernet\Teensy_WebSocketClientSocketIO_NativeEthernet.ino:126:
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSockets_Generic-Impl.h:74:6: warning: #warning No CORE_HAS_LIBB64 [-Wcpp]
     #warning No CORE_HAS_LIBB64
      ^
In file included from C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic.h:236:0,
                 from C:\Users\me\GitHub\Arduino\Teensy_WebSocketClientSocketIO_NativeEthernet\Teensy_WebSocketClientSocketIO_NativeEthernet.ino:126:
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic-Impl.h:303:4: warning: #warning WEBSOCKETS_NETWORK_TYPE != NETWORK_ESP8266_ASYNC [-Wcpp]
   #warning WEBSOCKETS_NETWORK_TYPE != NETWORK_ESP8266_ASYNC
    ^
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic-Impl.h:420:6: warning: #warning Not HAS_SSL [-Wcpp]
     #warning Not HAS_SSL
      ^
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic-Impl.h: In member function 'virtual void WebSocketsClient::clientDisconnect(WSclient_t*)':
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic-Impl.h:729:20: warning: deleting object of polymorphic class type 'EthernetClient' which has non-virtual destructor might cause undefined behaviour [-Wdelete-non-virtual-dtor]
     delete client->tcp;
                    ^
Opening Teensy Loader...
Memory Usage on Teensy 4.1:
  FLASH: code:235564, data:75072, headers:7276   free for files:7808552
   RAM1: variables:40100, code:95512, padding:2792   free for local variables:385884
   RAM2: variables:12416  free for malloc/new:511872
Teensy serial output:

Start Teensy_WebSocketClientSocketIO_NativeEthernet on TEENSY 4.1 with Custom Ethernet using Teensy 4.1 NativeEthernet Library
WebSockets_Generic v2.7.0
[WS] Board :TEENSY 4.1, setCsPin:10
[WS] Default SPI pinout:
[WS] MOSI:11
[WS] MISO:12
[WS] SCK:13
[WS] SS:10
[WS] =========================
WebSockets Client @ IP address: 192.168.1.117
Connecting to WebSockets Server @ IP address: 192.168.1.187, port: 3000
[WS] WebSockets_Generic v2.7.0
[WS] [wsIOc] found EIO=4 disable EIO ping on client
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
["event_name",{"now":45237}]
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
["event_name",{"now":77261}]
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
["event_name",{"now":109285}]
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
["event_name",{"now":141309}]
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
["event_name",{"now":173333}]
Ping:

ping 192.168.1.117

Pinging 192.168.1.117 with 32 bytes of data:
Reply from 192.168.1.117: bytes=32 time=1ms TTL=64
Reply from 192.168.1.117: bytes=32 time=1ms TTL=64
Reply from 192.168.1.117: bytes=32 time=1ms TTL=64
Reply from 192.168.1.117: bytes=32 time=1ms TTL=64

Ping statistics for 192.168.1.117:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 1ms, Maximum = 1ms, Average = 1ms
Wireshark
image

Screenshots
If applicable, add screenshots to help explain your problem.

Information
Please ensure to specify the following:

Arduino IDE version:1.8.15
TeensyDuino: 1.54 Beta9
Contextual information: looking to confirm packets are being sent
Anything that might be relevant in your opinion, such as:
Windows 10, firewall disabled MINGW32_NT-6.2 MPC 1.0.8(0.46/3/2) 2002-12-09 07:58 i686 unknown
Unifi network, Teensy hardwired Ethernet, PC is wifi

khoih-prog, 2021-06-22:
Hi @mechatroniks-git 

Thanks for your detailed report. I'm able to reproduce the issue and has a fix, which is a **bug in the NativeEthernet library.**

This is the debug terminal with the bug fix

```
Start Teensy_WebSocketClientSocketIO_NativeEthernet on TEENSY 4.1 with Custom Ethernet using Teensy 4.1 NativeEthernet Library
WebSockets_Generic v2.7.0
[WS] Board :TEENSY 4.1, setCsPin:10
[WS] Default SPI pinout:
[WS] MOSI:11
[WS] MISO:12
[WS] SCK:13
[WS] SS:10
[WS] =========================
WebSockets Client @ IP address: 192.168.2.103
Connecting to WebSockets Server @ IP address: 192.168.2.30, port: 8080
[WS] WebSockets_Generic v2.7.0
[WS] [wsIOc] found EIO=4 disable EIO ping on client
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] sendHeader: client->cKey = 1lTomU5X6SGDGrlASxqOlg==
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=polling HTTP/1.1
Host: 192.168.2.30:8080
Connection: keep-alive
Authorization: 1234567890
User-Agent: arduino-WebSocket-Client

[WS] [WS-Client] [sendHeader] Sending header... Done (us):18
Teensy4 Stream.cpp: HTTP/1.1 200 OK

[WS] [WS-Client][handleHeader] RX:HTTP/1.1 200 OK
Teensy4 Stream.cpp: Content-Type: text/plain; charset=UTF-8

[WS] [WS-Client][handleHeader] RX:Content-Type: text/plain; charset=UTF-8
Teensy4 Stream.cpp: Content-Length: 97

[WS] [WS-Client][handleHeader] RX:Content-Length: 97
Teensy4 Stream.cpp: Date: Tue, 22 Jun 2021 04:21:26 GMT

[WS] [WS-Client][handleHeader] RX:Date: Tue, 22 Jun 2021 04:21:26 GMT
Teensy4 Stream.cpp: Connection: keep-alive

[WS] [WS-Client][handleHeader] RX:Connection: keep-alive
Teensy4 Stream.cpp: 

[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=4
[WS] [WS-Client][handleHeader] - cKey:1lTomU5X6SGDGrlASxqOlg==
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:1
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:
[WS] [WS-Client][handleHeader] Still missing cSessionId try Socket.IO
[WS] [WS-Client][handleHeader] socket.io json: 0{"sid":"mJ7XjZxZ9y7mglEMAAAE","upgrades":["websocket"],"pingInterval":25000,"pingTimeout":20000}
[WS] [WS-Client][handleHeader] - cSessionId: mJ7XjZxZ9y7mglEMAAAE
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=4
[WS] [WS-Client][handleHeader] - cKey:1lTomU5X6SGDGrlASxqOlg==
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[... 119 lines cut ...]
    .cookie = this
  };

  fnet_dns_desc_t fnet_dns_desc = fnet_dns_init(&dns_params);
  
  if(fnet_dns_desc == FNET_NULL)
  {
    return 0;
  }
  
  while(resolveDone == 0)
  {
  }
  
  if(resolveDone == -1) 
    return 0;
  else if(resolveDone == 1) 
  {
  aResult = resolvedIP;
  return 1;
  }
    
  return 0 ;
}
```

closed 2021-06-22 by khoih-prog

khoih-prog, 2021-06-22:
The PR has been created for [**NativeEthernet** library](https://github.com/vjmuzik/NativeEthernet) as [**Fix the DNS bug** #21](https://github.com/vjmuzik/NativeEthernet/pull/21)

mechatroniks-git, 2021-06-29:
Hey @khoih-prog, I've updated the NativeEthernet  inside my Arduino folder (C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\NativeEthernet\src). I'm getting the same issue.

Compile warnings (are these expected?):
```
Teensy_WebSocketClientSocketIO_NativeEthernet:84: warning: #warning Using NativeEthernet lib for Teensy 4.1. Must also use Teensy Packages Patch or error 
   #warning Using NativeEthernet lib for Teensy 4.1. Must also use Teensy Packages Patch or error
    ^
In file included from C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic.h:56:0,
                 from C:\Users\me\GitHub\Arduino\Teensy_WebSocketClientSocketIO_NativeEthernet\Teensy_WebSocketClientSocketIO_NativeEthernet.ino:126:
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSockets_Generic.h:231:4: warning: #warning Use Teensy in WebSockets_Generic [-Wcpp]
   #warning Use Teensy in WebSockets_Generic
    ^
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSockets_Generic.h:516:4: warning: #warning Using Teensy 4.1 NativeEthernet Library [-Wcpp]
   #warning Using Teensy 4.1 NativeEthernet Library
    ^
In file included from C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSockets_Generic.h:718:0,
                 from C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic.h:56,
                 from C:\Users\me\GitHub\Arduino\Teensy_WebSocketClientSocketIO_NativeEthernet\Teensy_WebSocketClientSocketIO_NativeEthernet.ino:126:
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSockets_Generic-Impl.h:74:6: warning: #warning No CORE_HAS_LIBB64 [-Wcpp]
     #warning No CORE_HAS_LIBB64
      ^
In file included from C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic.h:236:0,
                 from C:\Users\me\GitHub\Arduino\Teensy_WebSocketClientSocketIO_NativeEthernet\Teensy_WebSocketClientSocketIO_NativeEthernet.ino:126:
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic-Impl.h:303:4: warning: #warning WEBSOCKETS_NETWORK_TYPE != NETWORK_ESP8266_ASYNC [-Wcpp]
   #warning WEBSOCKETS_NETWORK_TYPE != NETWORK_ESP8266_ASYNC
    ^
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic-Impl.h:420:6: warning: #warning Not HAS_SSL [-Wcpp]
     #warning Not HAS_SSL
      ^
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic-Impl.h: In member function 'virtual void WebSocketsClient::clientDisconnect(WSclient_t*)':
C:\Users\me\GitHub\Arduino\libraries\WebSockets_Generic\src/WebSocketsClient_Generic-Impl.h:729:20: warning: deleting object of polymorphic class type 'EthernetClient' which has non-virtual destructor might cause undefined behaviour [-Wdelete-non-virtual-dtor]
     delete client->tcp;
                    ^
Memory Usage on Teensy 4.1:
  FLASH: code:235756, data:75072, headers:8108   free for files:7807528
   RAM1: variables:40100, code:95704, padding:2600   free for local variables:385884
   RAM2: variables:12416  free for malloc/new:511872
```
Serial output
```
Start Teensy_WebSocketClientSocketIO_NativeEthernet on TEENSY 4.1 with Custom Ethernet using Teensy 4.1 NativeEthernet Library
WebSockets_Generic v2.7.0
[WS] Board :TEENSY 4.1, setCsPin:10
[WS] Default SPI pinout:
[WS] MOSI:11
[WS] MISO:12
[WS] SCK:13
[WS] SS:10
[WS] ========================|
WebSockets Client @ IP address: 192.168.1.119
Connecting to WebSockets Server @ IP address: 192.168.1.199, port: 8080
[WS] WebSockets_Generic v2.7.0
[WS] [wsIOc] found EIO=4 disable EIO ping on client
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
["event_name",{"now":49307}]
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
["event_name",{"now":89309}]
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
["event_name",{"now":129311}]
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
```

I setup a SocketIO server on 192.168.1.119 port 8080, firewall off. Used the 2nd server example from [here](https://python-socketio.readthedocs.io/en/latest/intro.html). Nothing coming out of the Teensy on wireshark, I do see the ping responses as before.

khoih-prog, 2021-06-29:
Hi,

Those warnings are intentional and correct.

I have no idea about what you're doing wrong or which steps /patches (Teensy core / NativeEthernet library / patches) you're missing as I have no problem at all as you can see the previously posted terminal output.

Teensy especially with **still not-yet-matured NativeEthernet** really creates many unsolved and hard-to-solve problems. I'm currently have no time and enough interest to investigate.

Be sure to check that these are installed correctly. I'm afraid I can help you any further with my crystal ball.

1. [Packages' Patches for Teensy boards](https://github.com/khoih-prog/WebSockets_Generic#2-for-teensy-boards)
2. Use this Stream.cpp to replace `./arduino-x.yy.zz/hardware/teensy/avr/cores/teensy4/Stream.cpp`

### Stream.h

```
/*
  Stream.h - base class for character-based streams.
  Copyright (c) 2010 David A. Mellis.  All right reserved.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#ifndef Stream_h
#define Stream_h

#include <inttypes.h>
#include "Print.h"

class Stream : public Print
{
  public:
    constexpr Stream() : _timeout(1000), read_error(0) {}
    virtual int available() = 0;
    virtual int read() = 0;
    virtual int peek() = 0;

    void setTimeout(unsigned long timeout);
    bool find(const char *target);
    
    bool find(const uint8_t *target) 
    {
      return find ((const char *)target);
    }
    
    bool find(const String &target) 
    {
      return find(target.c_str());
    }
    
    bool find(const char *target, size_t length);
    
    bool find(const uint8_t *target, size_t length) 
    {
      return find ((const char *)target, length);
    }
    
    bool find(const String &target, size_t length) 
[... 531 lines cut ...]

  while (length < max)
  {
    int c = timedRead();

    if (c < 0)
    {
      setReadError();
      break;  // timeout
    }

    if (c == 0 || c == terminator)
      break;

    str += (char)c;
    length++;
  }
  
  // MR fix
  if(str.length()==0) str="\0";

  return str;
}
#endif
```

khoih-prog, 2021-06-29:
I'm using this SIO server, included with the library 
[**Nodejs_SocketIO_TestServer**](https://github.com/khoih-prog/WebSockets_Generic/tree/master/examples/Nodejs_SocketIO_TestServer)

khoih-prog, 2021-06-29:
Just test with **Teensy core v1.54beta10 and Arduino v1.8.15** and it's still working well

#### NativeEthernet SIO Client

```
Start Teensy_WebSocketClientSocketIO_NativeEthernet on TEENSY 4.1 with Custom Ethernet using Teensy 4.1 NativeEthernet Library
WebSockets_Generic v2.7.0
[WS] Board :TEENSY 4.1, setCsPin:10
[WS] Default SPI pinout:
[WS] MOSI:11
[WS] MISO:12
[WS] SCK:13
[WS] SS:10
[WS] =========================
WebSockets Client @ IP address: 192.168.2.116
Connecting to WebSockets Server @ IP address: 192.168.2.30, port: 8080
[WS] WebSockets_Generic v2.7.0
[WS] [wsIOc] found EIO=4 disable EIO ping on client
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] sendHeader: client->cKey = JHxGz+EFdSxa6Mb9AFEzAg==
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=polling HTTP/1.1
Host: 192.168.2.30:8080
Connection: keep-alive
Authorization: 1234567890
User-Agent: arduino-WebSocket-Client

[WS] [WS-Client] [sendHeader] Sending header... Done (us):19
[WS] [WS-Client][handleHeader] RX:HTTP/1.1 200 OK
[WS] [WS-Client][handleHeader] RX:Content-Type: text/plain; charset=UTF-8
[WS] [WS-Client][handleHeader] RX:Content-Length: 97
[WS] [WS-Client][handleHeader] RX:Date: Tue, 29 Jun 2021 22:01:47 GMT
[WS] [WS-Client][handleHeader] RX:Connection: keep-alive
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=4
[WS] [WS-Client][handleHeader] - cKey:JHxGz+EFdSxa6Mb9AFEzAg==
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:1
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:
[WS] [WS-Client][handleHeader] Still missing cSessionId try Socket.IO
[WS] [WS-Client][handleHeader] socket.io json: 0{"sid":"u6mgnv5wm5LrN0leAAAA","upgrades":["websocket"],"pingInterval":25000,"pingTimeout":20000}
[WS] [WS-Client][handleHeader] - cSessionId: u6mgnv5wm5LrN0leAAAA
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=4
[WS] [WS-Client][handleHeader] - cKey:JHxGz+EFdSxa6Mb9AFEzAg==
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:1
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:u6mgnv5wm5LrN0leAAAA
[WS] [WS-Client][handleHeader] found cSessionId
[WS] [WS-Client] [sendHeader] Sending header...
[WS] sendHeader: client->cKey = D2lVAT+6DJTL652keZxVIw==
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=websocket&sid=u6mgnv5wm5LrN0leAAAA HTTP/1.1
Host: 192.168.2.30:8080
Connection: Upgrade
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: D2lVAT+6DJTL652keZxVIw==
Sec-WebSocket-Protocol: arduino
Authorization: 1234567890
User-Agent: arduino-WebSocket-Client

[WS] [WS-Client] [sendHeader] Sending header... Done (us):26
[WS] [WS-Client][handleHeader] RX:HTTP/1.1 101 Switching Protocols
[WS] [WS-Client][handleHeader] RX:Upgrade: websocket
[WS] [WS-Client][handleHeader] RX:Connection: Upgrade
[WS] [WS-Client][handleHeader] RX:Sec-WebSocket-Accept: T24rdxNb6xZZ51NYE8YDXiB/mYE=
[WS] [WS-Client][handleHeader] RX:Sec-WebSocket-Protocol: arduino
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=4
[WS] [WS-Client][handleHeader] - cKey:D2lVAT+6DJTL652keZxVIw==
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:101
[WS] [WS-Client][handleHeader] - cIsUpgrade:1
[WS] [WS-Client][handleHeader] - cIsWebsocket:1
[WS] [WS-Client][handleHeader] - cAccept:T24rdxNb6xZZ51NYE8YDXiB/mYE=
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:u6mgnv5wm5LrN0leAAAA
[WS] [WS-Client][handleHeader] Websocket connection init done.
[WS] [wsIOc] Connected to url:/socket.io/?EIO=4
[IOc] Connected to url: /socket.io/?EIO=4
[WS] [wsIOc] get pong
[WS] [wsIOc] connected: len = 30
[WS] [wsIOc] data: {"sid":"P7Isy2B2k8G0uLHTAAAB"}

```
---

#### Server

```
Server launched on port 8080
Connected
P7Isy2B2k8G0uLHTAAAB
JWT token test:  { host: '192.168.2.30:8080',
  connection: 'keep-alive',
  authorization: '1234567890',
  'user-agent': 'arduino-WebSocket-Client' }
Message from Client :  { now: 30001 }
Message from Client :  { now: 60002 }
Message from Client :  { now: 90003 }

```

khoih-prog, 2021-06-29:
You also can print more debugging messages by using in the sketch

```
#define _WEBSOCKETS_LOGLEVEL_     4
```

mechatroniks-git, 2021-06-29:
None of that worked. 

I'm looking at my router logs now, I see the DHCP:

```
Jun 29 22:32:54 ubnt daemon.info dnsmasq-dhcp[2883]: DHCPDISCOVER(br0) de:ad:be:ef:fe:01
Jun 29 22:32:54 ubnt daemon.info dnsmasq-dhcp[2883]: DHCPOFFER(br0) 192.168.1.73 de:ad:be:ef:fe:01
Jun 29 22:32:54 ubnt daemon.info dnsmasq-dhcp[2883]: DHCPREQUEST(br0) 192.168.1.73 de:ad:be:ef:fe:01
Jun 29 22:32:54 ubnt daemon.info dnsmasq-dhcp[2883]: DHCPACK(br0) 192.168.1.73 de:ad:be:ef:fe:01
```

khoih-prog, 2021-06-29:
[Teensy_WebSocketClientSocketIO_NativeEthernet.ino.hex.zip](https://github.com/khoih-prog/WebSockets_Generic/files/6736980/Teensy_WebSocketClientSocketIO_NativeEthernet.ino.hex.zip)

khoih-prog, 2021-06-29:
Are you sure your SIO server is working OK? Can't you test it using a simple SIO client to be sure not wasting your time.

mechatroniks-git, 2021-06-29:
Thanks for the hex file, same thing. I will try setting up SIO server on a raspberry pi next.

khoih-prog, 2021-06-29:
I'm just run that [**Nodejs_SocketIO_TestServer**](https://github.com/khoih-prog/WebSockets_Generic/tree/master/examples/Nodejs_SocketIO_TestServer) on my Ubuntu PC/Laptop, same code will work on RPi (with Nodejs)

mechatroniks-git, 2021-06-30:
Ok, @khoih-prog  I setup the PI with that TestServer (192.168.1.9) and my python client (192.168.1.199) can connect, I see packets on wireshark. 

[image]

Still nothing from the Teensy 4.1, I tried 2nd set of hardware that works fine with NativeEthernet sending UDP packets all day long. 

I'm using the Teensy 4.1 with onboard Ethernet PHY chip, which is NOT connected via SPI, it has a RMII interface to the IMX chip. The PHY connects to Ethernet with one of [these](https://www.pjrc.com/store/ethernet_kit.html). Can you confirm your setup is the same?

Thanks.

khoih-prog, 2021-06-30:
Same Ethernet MagJack kit. 

I suggest you have to recheck everything individually (network, server, Teensy, NativeEthernet, etc.) to be sure everything is OK before moving forward. I'm thinking there is something wrong with your NativeEthernet (MagJack kit, wiring, solder, Teensy, etc.). Only you can figure this out, sorry.

[image]

mechatroniks-git, 2021-06-30:
@khoih-prog Thanks, you've been very helpful. I will keep at it, setup a dumb network, I'm thinking this Unifi UDM system is blocking packets.

khoih-prog, 2021-09-06:
Hi @mechatroniks-git 

FYI, you can also test and use this new release, using the new and much better [**QNEthernet** Library](https://github.com/ssilverman/QNEthernet) for Teensy 4.1 built-in Ethernet

---

### Major Releases v2.9.0

1. Add support to [**QNEthernet** Library](https://github.com/ssilverman/QNEthernet) for Teensy 4.1 built-in Ethernet
2. Update examples with new features

---

### [Teensy_WebSocketClientSocketIO_QNEthernet](https://github.com/khoih-prog/WebSockets_Generic/tree/master/examples/Teensy/QNEthernet/Teensy_WebSocketClientSocketIO_QNEthernet) on TEENSY 4.1 using QNEthernet

The following are debug terminal output when running example [Teensy_WebSocketClientSocketIO_QNEthernet](https://github.com/khoih-prog/WebSockets_Generic/tree/master/examples/Teensy/QNEthernet/Teensy_WebSocketClientSocketIO_QNEthernet) on TEENSY 4.1 using [`QNEthernet`](https://github.com/ssilverman/QNEthernet) library

### Client

```
Starting Teensy_WebSocketClient_QNEthernet on TEENSY 4.1 using QNEthernet
WebSockets_Generic v2.9.0
=========== USE_QN_ETHERNET ===========
Initialize Ethernet using static IP => Connected! IP address:192.168.2.222
Connecting to WebSockets Server @ IP address: 192.168.2.30, port: 8080
[IOc] Connected to url: /socket.io/?EIO=4
["event_name",{"now":30001}]
[IOc] Get event: ["Send Message io.emit Broadcasted : ",{"now":30001}]
[IOc] Get event: ["Send Message : ",{"now":30001}]
["event_name",{"now":60002}]
[IOc] Get event: ["Send Message io.emit Broadcasted : ",{"now":60002}]
[IOc] Get event: ["Send Message : ",{"now":60002}]
["event_name",{"now":90003}]
[IOc] Get event: ["Send Message io.emit Broadcasted : ",{"now":90003}]
[IOc] Get event: ["Send Message : ",{"now":90003}]
```

---

####  Server

Using the Socket.IO Server in [**Nodejs_SocketIO_TestServer**](examples/Nodejs_SocketIO_TestServer)

```
kh@kh-Inspiron-3593:~/Downloads/Nodejs/New_SIO$ ./SIO_server.sh 
Server launched on port 8080
Connected
bK733pVgbd3DCsy5AAAB
JWT token test:  { host: '192.168.2.30:8080',
  connection: 'keep-alive',
  authorization: '1234567890',
  'user-agent': 'arduino-WebSocket-Client' }
Message from Client :  { now: 30001 }
Message from Client :  { now: 60002 }
Message from Client :  { now: 90003 }
```