khoih-prog/WebSockets_Generic #13: Loss of connection of the arduino MKR with the nodejs server when the nodejs server is restarted.

Josua2012, 2021-10-07:
Hello Khoih,

We haven't spoken for a long time, I hope you are feeling very well.

I have a serious problem with socket.io, as you know I am using nodejs as a server and an Arduino MKR1000.
I have observed the following bug, when I turn off the nodejs and then run it, an automatic reconnection of the arduinos to the nodejs generally occurs, however, this does not always happen, on many occasions, the arduino does not reconnect with the nodejs server and I have no choice but to turn the arduinos power off and turn them back power on.
I also have a web application created with Flutter and that uses socket.io that does not have this problem, so it is obviously a problem with the websockets_generic library.
Would you have any idea how to solve this problem? 

Thank you very much for your help. 

Regards,
Jose A.Molina

khoih-prog, 2021-10-07:
Hi @Josua2012 

Thanks for reporting this serious bug.

I'll spend some time later to investigate and fix the issue, as I'm currently busy working on [**AsyncWebserver for Portenta H7** #6](https://github.com/khoih-prog/AsyncWebServer_STM32/discussions/6)

I'd appreciate if you now can spend more time to investigate, test and/or help pointing the direction to narrow down the cause.

Best,

jamolina2021, 2021-10-08:
Hi @khoih-prog

I have observed that if the NodeJS server is in Windows 10 the problem does not occur.
The problem occurs if the **NodeJS server is on linux ubuntu 20.04**

The files of the nodejs project are the following:
[app.zip](https://github.com/khoih-prog/WebSockets_Generic/files/7309433/app.zip)

The sketch is for an ethernet connection and is as follows:
[SketchPruebaSocket.zip](https://github.com/khoih-prog/WebSockets_Generic/files/7309444/SketchPruebaSocket.zip)

Next I send you the debug output terminal:
[debug.zip](https://github.com/khoih-prog/WebSockets_Generic/files/7309529/debug.zip)

The connections that I have made until the problem has occurred have been the following:

```
HN1CluOsT7bsqpxFAAAB
8ozcx9GcS6bSnY-wAAAB
_XA1qPp9tELFRfCRAAAB
KTUCwjsodRo6A2QnAAAB
eHrZ6fSbJF8aWFcrAAAB
P123uIaisE8enw6SAAAB
xBA4g9p2qHEo_SL5AAAB
tPirFzu1EjZOq249AAAB
```

As you can see, after connecting sid: **tPirFzu1EjZOq249AAAB**, shutdown node server and run **node app** again the arduino could no longer connect to the server.

The steps to follow to try to reproduce the problem are:
1) Connect arduino MKR1000
2) Connect nodejs server (node app) in linux terminal window
3) CTRL + C in the terminal window shutdown the nodejs server
4) node app (the arduino can connect to the server)
5) CTRL + C (server shutdown)
6) node app (the arduino can connect to the server)
7) CTRL + C (server shutdown)
....
10) Repeat previous steps until the arduino can no longer connect to the server by doing: node app 

Tell me if these investigations have helped you. 

Thanks for your help.

khoih-prog, 2021-10-08:
> The strange thing is that if it is a NodeJS problem in Ubuntu when the connection is via WiFi it should not work correctly either, however, it works perfectly.

Sometimes, there is just some `feature` affecting certain software, for example, [**Fail upload sketches to ESP8266 only on Ubunto 20.04.3**](https://arduino.stackexchange.com/questions/86211/fail-upload-sketches-to-esp8266-only-on-ubunto-20-04-3) affecting only ESP8266 uploading process while it's OK for other boards' cores

> I am going to buy a DFRobot W5500 Ethernet card with POE control board (compatible with Arduino) to see if it works correctly .
> Is this card supposed to be compatible with WebSockets_generic library?

If it's W5500, 99.99% will be supported directly. Did you try using better `Ethernet` libraries for **MKR ETH Shield**, such as `Ethernet2, Ethernet3 or EthernetLarge` ? If not, try again to see if it's any better.

> Regarding the article that you comment, from what I see, it only talks about a problem of uploading sketches to ESP8266 in Ubuntu, but I am not using Ubuntu to upload sketches, Ubuntu I am only using it to host the server created with nodejs.

Just an example for you to know different kernel can introduce some unexpected issue, we don't know until verify / experience. That's why changing the kernel is one step to isolate the issue : kernel-related or not?

Good Luck,

Josua2012, 2021-10-08:
> These 2K buffer limitations are fixed by Libraries' Patches unless there are some limitation introduced by the MKR ETH board hardware or your version of Ethernet library

The question about the limitations in the sending referred to WiFi101, not to Ethernet, sorry not to have specified it before.
Can this limit be increased using WiFi101?

khoih-prog, 2021-10-11:
Did you install [**EthernetLarge** library](https://github.com/OPEnSLab-OSU/EthernetLarge) ?

```
#define USE_ETHERNET_LARGE          true
#include <EthernetLarge.h>
#warning Using EthernetLarge lib
#define SHIELD_TYPE           "W5x00 using EthernetLarge Library"
```

I'm afraid you're moving too fast. Try going back from beginning point of any Ethernet Library, and use an example of that library and see if there is any issue, then moving on to more complex libraries and use-cases.

I compile using either `Ethernet2`, `Ethernet3`, `WebSockets_Generic` for MKR1000 without any error at all, and **I'm using all the patches.**

So check there is possibly something wrong with your installation, file corruption, outdated version, etc.

---

### Ethernet2

[**WebClient** example](https://github.com/khoih-prog/Ethernet2/tree/master/examples/WebClient)

[image]

---

### Ethernet3

[**WebClient** example](https://github.com/khoih-prog/Ethernet3/tree/master/examples/WebClient)

[image]

---

### WebSockets_Generic using EthernetLarge

[**Generic_WebSocketClientSocketIO_W5500** example](https://github.com/khoih-prog/WebSockets_Generic/tree/master/examples/Generic/W5500/Generic_WebSocketClientSocketIO_W5500)

[image]

jamolina2021, 2021-10-13:
> Just remember I had a patch for WiFi101 Library as required in my WiFiWebServer Library's Prerequisites

Yes, when we were looking at the compatibility of websockets_generic with socket.io v.3 you had to modify the WiFi101 library for it to work correctly.
It works perfectly for me, except for the 2K limitation in shipping. 

I have been able to solve the problem of the installation of the "ethernet" library and I have been able to include the patch, but with the patch it cannot connect to the node server.

I have made new tests with all the ethernet libraries as you indicated and my conclusions are the following in case anyone else asks you about this type of problem: 

**Ethernet**
Connect only if USE_UIP_ETHERNET true
If the patch is put, it compiles correctly but does not connect with the server
Without the patch it compiles correctly and connects correctly with the server
But you still have the problem that sometimes when the Node server is restarted it won't reconnect

**EthernetLarge**
Connect only if USE_UIP_ETHERNET true
Compile correctly with or without patch
Connects correctly to the server with patch or without patch
But you still have the problem that sometimes when the Node server is restarted it won't reconnect

**Ethernet2**
Without the patch it compiles without errors but does not connect with the server
With the patch it compiles without errors but does not connect to the server either
The WebClient sample compiles without errors but does not connect.

**Ethernet3**
Without the patch it compiles without errors but does not connect with the server
With the patch it compiles without errors but does not connect to the server either 
The WebClient sample compiles without errors but does not connect.

Apparently Ethernet2 and Ethernet3 have some incompatibility with MKR1000 + MKR ETH Shield.

So there must be some other circumstance where the reconnection with the nodejs server does not occur in all circumstances (Ethernet and EthernetLarge).
And maybe the problem has nothing to do with the Ethernet libraries ...

 Again, thank you very much for all your help.

jamolina2021, 2021-10-14:
Hi @khoih-prog 
I already have the **DFRobot W5500 board**, from what I have read I have to select the **Arduino Leonardo** in the Arduino Studio, but I have a question regarding your sketch.
In this part of your code, what do I have to put to make it compatible with the **Arduino Leonardo**?

```
#if ( defined(ARDUINO_SAM_DUE) || defined(__SAM3X8E__) )
  // Default pin 10 to SS/CS
  #define USE_THIS_SS_PIN       10
  #define BOARD_TYPE      "SAM DUE"
#endif

#ifndef BOARD_NAME
  #define BOARD_NAME    BOARD_TYPE
#endif
```

In the instructions of this card it indicates that I should put at the beginning of the sketch this:

```
#define SS     10U    //D10 ---- SS
#define RST    11U    //D11 -----Reset
```

Thank you very much for all your help.