khoih-prog/WebSockets_Generic #14: Problem with connecting to a nodejs (socket.io)

Josua2012, 2021-11-10:
Hello Khoih,

The library must have a problem.
Remember that I told you that sometimes, when restarting the servers, the mkr 1000 with ethernet shield and with your library, socket.io could not reconnect with the server, and the only way to make it work again was to turn off the arduino and restart it.
You told me that the problem was the fault of nodejs on ubuntu.
But now I am using the mkr1000 over wifi (wifi101) and the nodejs server is on Windows and I have exactly the same problem, so undoubtedly, the library must have some problem, and in certain circumstances it cannot reconnect with the server.
This is a serious problem and makes your library unusable in an environment that requires connection security.
The difficulty here is the reproduction of this problem, as it occurs at some point, it can happen a few hours or occur immediately, sometimes restarting the server, other times without doing anything special. 
It's very complicated.
Do you have any idea to solve it? 

Thanks for the help.
Hope you're well. 

Regards,
Jose.

khoih-prog, 2021-11-10:
Hi,

> You told me that the problem was the fault of nodejs on ubuntu.

This is the guessing **false alarm**, please ignore it. The real culprit is the USB driver of certain Ubuntu releases and has been fixed with the latest kernels.

> the mkr 1000 with ethernet shield

This combination is not the good enough and popular choice. Using the popular boards has **many advantages**, such as price, less bug as they are identified and fixed if any, easy sourcing, better choice of libraries, etc.

Try with other good boards (ESP32, ESP8266, Nano_33_IoT, Nano_RP2040_Connect, RP2040, nRF52, etc.)  to see if this erratic problem is board (MKR1000) dependent / Ethernet library dependent. 

Check [**Currently supported Boards**](https://github.com/khoih-prog/WebSockets_Generic#currently-supported-boards)

As you possibly know, this library has to rely on too many underlying layers, such as **core, hardware, ethernet/wifi library, websocket server, etc**. and every of them must be perfect to ensure the reliable operation of this library.

Certainly nobody can investigate / trace and debug the issue but you.

I can't help anything as I don't have the board combination / can't duplicate the issue and won't spend time until a bug of this library is proven.

Good Luck,

closed 2021-11-10 by khoih-prog

Josua2012, 2021-11-10:
Thanks @Khoih-prog