khoih-prog/WebSockets_Generic #15: Compile error on Wio Terminal: 'class WiFiClientSecure' has no member named 'setInsecure'

RoSchmi, 2021-11-10:
Hi,
I'm just updating a 'Sinric Pro' application for the Wio Terminal from the lib SinricPro_Generic v2.8.0 to v2.8.3 and now get the compilation error from the included WebSockets_Generic library:
.pio\libdeps\SAMD\WebSockets_Generic@src-3e25c04322622ed35619bd8f1cb33dbd\src/WebSocketsClient_Generic-Impl.h:396:22: error: 'class WiFiClientSecure' has no member named 
'setInsecure'
         _client.ssl->setInsecure();

It's not a big thing and really a problem for me, but I think that it should not occur and I only wanted to inform about the issue.

PlatformIo Home v3.4.0
Board: Seeed Wio Terminal

Btw: Again many thanks for your valuable work. Lately I used your 'ESPAsync_WiFiManager' in an application (https://github.com/RoSchmi/Esp32_WiFiManager_HeatingSurvey) and it works like a charm.

Best regards
RoSchmi

khoih-prog, 2021-11-11:
Hi @RoSchmi 

Thanks for your report, I definitely will have a look at the issue soon as I'm busy with other libraries now. If possible, could you post an [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example) to help me start.

I'm just guessing the Wio Terminal WiFi library is not ready for that `setInsecure()` feature as I haven't touched WIO Terminal quite a while.

BTW, I'm really impressed with your recent project. So **complex and useful**

Best Regards,

khoih-prog, 2021-11-11:
HI,

I can duplicate the issue now. Will fix ASAP

BR,

khoih-prog, 2021-11-11:
Hi @RoSchmi 

I don't think I'll spend time on this WIO Terminal issue because the current status of Seeed libraries is terribly buggy  !!!

Just try to compile libraries' examples for WIO and verify. For example:

1. [wifi_sta](https://github.com/Seeed-Studio/Seeed_Arduino_rpcUnified/tree/master/examples/wifi_sta)
2. [WiFiClientSecure](https://github.com/Seeed-Studio/Seeed_Arduino_rpcWiFi/tree/master/examples/WiFiClientSecure)

and see you'll receive a bunch of errors, even with latest 

1. [**Seeed_Arduino_rpcUnified** releases v2.1.3](https://github.com/Seeed-Studio/Seeed_Arduino_rpcUnified/releases/tag/v2.1.3)
2. [**Seeed_Arduino_rpcWiFi** releases v1.0.5](https://github.com/Seeed-Studio/Seeed_Arduino_rpcWiFi/releases/tag/v1.0.5)

I suggest if you're still interested, please test and verify which version is OK to use for each of Seeed libraries. I don't have patience to do this anymore. Sorry

BR,

---

PS: To remove the `_client.ssl->setInsecure()` error, just change [**WebSocketsClient_Generic-Impl.h**#L392](https://github.com/khoih-prog/WebSockets_Generic/blob/master/src/WebSocketsClient_Generic-Impl.h#L392) to

```
#if defined(ESP32) //|| (WEBSOCKETS_NETWORK_TYPE == NETWORK_RTL8720DN)
```

 from

```
#if defined(ESP32) || (WEBSOCKETS_NETWORK_TYPE == NETWORK_RTL8720DN)
```

closed 2021-11-11 by khoih-prog

RoSchmi, 2021-11-11:
Yes, you are right. I didn't use the Wio Terminal for new projects for half a year and I wanted to actualize some of my projects on the latest WiFi-module firmware to see if they have solved the issues. Unfortunately the first tested application (for Sinric Pro) regulary hangs after an hour or so. So I will switch to Esp32 which seems to work reliably as far as I can say for now.
It's a pity that apparently seeed doesn't get the networking/internet stuff to work reliably for such a nice board.
The compiling error of your lib isn't a problem.
Thanks for your advice
RoSchmi