khoih-prog/WebSockets_Generic #24: Arduino Portenta not connect to SocketIO example

elastra21, 2022-05-13:
### Arduino Portenta not connect to SocketIO example

I'm trying to connect a Arduino Portenta as SocketIO client to the SocketIO server example in this repo and it keeps disconnecting. After a few tries a check if the problem was the portenta or the library so I ran the same example with a ESP-32 and a ESP-8266 and both works but when I try with Portenta keep Disconnecting and it's weird becuase in the Server logs never shows a connection but with ESP32 or ESP8266 shows each time that a client is Connected.

### Steps to Reproduce

- Arduino mbed OS portenta boards V3.0.1
- WebSocketClientSocketIO_WiFi.ino ------> Client Example
- https://github.com/khoih-prog/WebSockets_Generic/tree/master/examples/Nodejs_SocketIO_TestServer  ------> Server Example

### Information

Please ensure to specify the following:

* Arduino IDE Version: 2.0.0-rc6
* `Arduino Portenta`
* Arduino mbed OS portenta boards V3.0.1
  
### Example

```
Start WebSocketClientSocketIO_WiFi on PORTENTA_H7_M7WebSockets_Generic v2.15.0
Connecting to SSID: Lastra
SSID: Lastra
WebSockets Client IP Address: 192.168.0.107
signal strength (RSSI):-62 dBm
Connecting to WebSockets Server @ IP address: 192.168.0.5, port: 8080
[WS] [WS-Client] beginSocketIO with IPAddress
[WS] [WS-Client] beginSocketIO with const char
[WS] WebSockets_Generic v2.15.0
[WS] [wsIOc] found EIO=4 disable EIO ping on client
[WS] [WS-Client] connectedCb
[WS] [WS-Client][connectedCb] Connected to Host:192.168.0.5, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] sendHeader: client->cKey = mrl30JBiab+xRJYPbSZE6A==
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=polling HTTP/1.1
Host: 192.168.0.5:8080
Connection: keep-alive
Authorization: 1234567890
User-Agent: arduino-WebSocket-Client

[WS] [write] n:165, t:20558
[WS] [write] Write, Length :165, Left :0
[WS] [WS-Client] [sendHeader] Sending header... Done (us):1110
[WS] [WS-Client][handleHeader] RX:HTTP/1.1 200 OK
[WS] [WS-Client][handleHeader] RX:Content-Type: text/plain; charset=UTF-8
[WS] [WS-Client][handleHeader] RX:Content-Length: 118
[WS] [WS-Client][handleHeader] RX:Date: Fri, 13 May 2022 19:16:21 GMT
[WS] [WS-Client][handleHeader] RX:Connection: keep-alive
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=4
[WS] [WS-Client][handleHeader] - cKey:mrl30JBiab+xRJYPbSZE6A==
[WS] [WS-Client][handleHeader] Server header:

[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:1
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:
[WS] [WS-Client][handleHeader] Still missing cSessionId try Socket.IO
[WS] [WS-Client][handleHeader] socket.io json: 0{"sid":"uHxZNe6cfsiJuPMGAAAL","upgrades":["websocket"],"pingInterval":25000,"pingTimeout":20000,"maxPayload":1000
[WS] [WS-Client][handleHeader] - cSessionId: uHxZNe6cfsiJuPMGAAAL
[WS] [WS-Client][handleHeader] Header read fin.
[WS] [WS-Client][handleHeader] Client settings:
[WS] [WS-Client][handleHeader] - cURL:/socket.io/?EIO=4
[WS] [WS-Client][handleHeader] - cKey:mrl30JBiab+xRJYPbSZE6A==
[WS] [WS-Client][handleHeader] Server header:
[WS] [WS-Client][handleHeader] - cCode:200
[WS] [WS-Client][handleHeader] - cIsUpgrade:0
[WS] [WS-Client][handleHeader] - cIsWebsocket:1
[WS] [WS-Client][handleHeader] - cAccept:
[WS] [WS-Client][handleHeader] - cProtocol:arduino
[WS] [WS-Client][handleHeader] - cExtensions:
[WS] [WS-Client][handleHeader] - cVersion:0
[WS] [WS-Client][handleHeader] - cSessionId:uHxZNe6cfsiJuPMGAAAL
[WS] [WS-Client][handleHeader] found cSessionId
[WS] [WS-Client] [sendHeader] Sending header...
[WS] sendHeader: client->cKey = DTT5PkIBNTgxX3MUKD/XYg==
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=websocket&sid=uHxZNe6cfsiJuPMGAAAL HTTP/1.1
Host: 192.168.0.5:8080
Connection: Upgrade
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: DTT5PkIBNTgxX3MUKD/XYg==
Sec-WebSocket-Protocol: arduino
Authorization: 1234567890
User-Agent: arduino-WebSocket-Client

[WS] [write] n:314, t:20585
[WS] [write] Write, Length :314, Left :0
[WS] [WS-Client] [sendHeader] Sending header... Done (us):1368
[WS] [WS-Client][handleHeader] RX:000}HTTP/1.1 101 Switching Protocols
[WS] [WS-Client][handleHeader] Header error:000}HTTP/1.1 101 Switching Protocols
[WS] [WS-Client][handleHeader] RX:Upgrade: websocket
[WS] [WS-Client][handleClientData] Header response timeout.. Disconnecting!
[WS] [WS-Client] client disconnected.
[IOc] Disconnected
[WS] [wsIOc] Disconnected!
["event_name",{"now":30001}]
[WS] [WS-Client] connectedCb
[WS] [WS-Client][connectedCb] Connected to Host:192.168.0.5, Port:8080
[WS] [WS-Client] [sendHeader] Sending header...
[WS] sendHeader: client->cKey = ZqaYvn9TscOSf2OKSewQmg==
[WS] [WS-Client] [sendHeader] Handshake:GET /socket.io/?EIO=4&transport=polling HTTP/1.1
Host: 192.168.0.5:8080
Connection: keep-alive
Authorization: 1234567890
User-Agent: arduino-WebSocket-Client

```

khoih-prog, 2022-05-13:
> Arduino IDE Version: 2.0.0-rc6

Try using **Arduino IDE v1.8.19**, not the **beta-and-buggy v2.0.0-rcx**

Its was OK here [**WebSocketClientSocketIO_WiFi on Portenta_H7 using WiFi**](https://github.com/khoih-prog/WebSockets_Generic#14-websocketclientsocketio_wifi-on-portenta_h7-using-wifi)

closed 2022-05-13 by khoih-prog

khoih-prog, 2022-05-13:
FYI, just tested using Arduino IDE v1.8.19 and still OK now.

```
Start WebSocketClientSocketIO_WiFi on PORTENTA_H7_M7
WebSockets_Generic v2.15.0
Connecting to SSID: HueNet1
SSID: HueNet1
WebSockets Client IP Address: 192.168.2.119
signal strength (RSSI):-29 dBm
Connecting to WebSockets Server @ IP address: 192.168.2.30, port: 8080
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WS] [wsIOc] Connected to url:/socket.io/?EIO=4
[IOc] Connected to url: /socket.io/?EIO=4
[WS] [wsIOc] get pong
[IOc] Get PONG
[WS] [wsIOc] connected: len = 30
[WS] [wsIOc] data: {"sid":"dqs_yYqGre5KIQ1QAAAB"}
["event_name",{"now":30001}]
[WS] [wsIOc] get event: len = 53
[WS] [wsIOc] get data: ["Send Message io.emit Broadcasted : ",{"now":30001}]
[IOc] Get event: ["Send Message io.emit Broadcasted : ",{"now":30001}]
[WS] [wsIOc] get event: len = 33
[WS] [wsIOc] get data: ["Send Message : ",{"now":30001}]
[IOc] Get event: ["Send Message : ",{"now":30001}]
[WS] [wsIOc] get ping send pong:3
[IOc] Get PING
```

```
Server launched on port 8080
Connected
dqs_yYqGre5KIQ1QAAAB
JWT token test:  {
  host: '192.168.2.30:8080',
  connection: 'keep-alive',
  authorization: '1234567890',
  'user-agent': 'arduino-WebSocket-Client'
}
Message from Client :  { now: 30001 }
Message from Client :  { now: 60002 }
Message from Client :  { now: 90003 }
Message from Client :  { now: 120004 }
```

elastra21, 2022-05-14:
Thanks for responding, I tried. with the Arduino v1.8.19 to see if works but no, I'm still getting the same error. I don't know what I'm doing wrong. I downloaded the example Server and tested with ESP32 and ESP8266 without problems but this one it's driving me crazy.

[image]
?

khoih-prog, 2022-05-14:
It's possible that you haven't applied the patch mentioned in [**[Portenta_H7] WiFi WebServer extremely slow from core v2.7.2 - v3.0.1** #441](https://github.com/arduino/ArduinoCore-mbed/issues/441)

Try to use the fix in [**Drastically speedup client.read() operation** #439](https://github.com/arduino/ArduinoCore-mbed/pull/439) by changing in these 2 files

1. [**MbedClient.cpp**](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/SocketWrapper/src/MbedClient.cpp)
2. [**MbedClient.h**](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/SocketWrapper/src/MbedClient.h)

**It's very bad that Arduino is so slow to fix the worst and breaking bugs in the core by releasing a new core version**

---
---

1. [**MbedClient.cpp**](https://github.com/arduino/ArduinoCore-mbed/blob/master/libraries/SocketWrapper/src/MbedClient.cpp)

```
#include "MbedClient.h"

#ifndef SOCKET_TIMEOUT
#define SOCKET_TIMEOUT 1500
#endif

arduino::MbedClient::MbedClient()
  : _status(false),
    _timeout(0) {
}

uint8_t arduino::MbedClient::status() {
  return _status;
}

void arduino::MbedClient::readSocket() {
  while (1) {
    event->wait_any(0xFF, 100);
    uint8_t data[SOCKET_BUFFER_SIZE];
    int ret = NSAPI_ERROR_WOULD_BLOCK;
    do {
      if (rxBuffer.availableForStore() == 0) {
        yield();
        continue;
      }
      mutex->lock();
      if (sock == nullptr || (closing && borrowed_socket)) {
        goto cleanup;
      }
      ret = sock->recv(data, rxBuffer.availableForStore());
      if (ret < 0 && ret != NSAPI_ERROR_WOULD_BLOCK) {
        goto cleanup;
      }
      if (ret == NSAPI_ERROR_WOULD_BLOCK || ret == 0) {
        yield();
        mutex->unlock();
        continue;
      }
      for (int i = 0; i < ret; i++) {
        rxBuffer.store_char(data[i]);
      }
      mutex->unlock();
      _status = true;
    } while (ret == NSAPI_ERROR_WOULD_BLOCK || ret > 0);
  }
cleanup:
  _status = false;
  return;
}

void arduino::MbedClient::getStatus() {
  event->set(1);
}

void arduino::MbedClient::setSocket(Socket *_sock) {
[... 371 lines cut ...]
  void onBeforeConnect(mbed::Callback<int(void)> cb) {
    beforeConnect = cb;
  }

private:
  RingBufferN<SOCKET_BUFFER_SIZE> rxBuffer;
  bool _status = false;
  bool borrowed_socket = false;
  bool _own_socket = false;
  bool closing = false;
  mbed::Callback<int(void)> beforeConnect;
  SocketAddress address;
  rtos::Thread* reader_th = nullptr;
  rtos::EventFlags* event = nullptr;
  rtos::Mutex* mutex = nullptr;
  unsigned long _timeout;

  void readSocket();
  void getStatus();
};

}

#endif
```

khoih-prog, 2022-05-14:
Also check the following Arduino Forum thread

[**Portenta H7 WebServer is very slowly**](https://forum.arduino.cc/t/portenta-h7-webserver-is-very-slowly/980076)

khoih-prog, 2022-05-14:
Be sure that you're using [**WiFiWebServer** releases v1.8.0](https://github.com/khoih-prog/WiFiWebServer/releases/tag/v1.8.0)

khoih-prog, 2022-05-17:
Good news.

> **It's very bad that Arduino is so slow to fix the worst and breaking bugs in the core by releasing a new core version**

The new [**ArduinoCore-mbed releases v3.1.1**](https://github.com/arduino/ArduinoCore-mbed/releases/tag/3.1.1) has just been released to fix the issues.

You can use that new release without manually applying the patch yourself.

elastra21, 2022-05-17:
Thank you verry much for the support, it works with the ArduinoCore-mbed v3.1.1, I didn't tried the the manual patch but the new release work perfect. 

And again thank you so much for your time.