khoih-prog/WebSockets_Generic #26: Nano IOT and platform IO "Communication with WiFi module failed!"

aldricGit, 2022-10-14:
Hello, 
When running any example on different Nano IOT I noticed that when using platform io you always end up with the message "Communication with WiFi module failed!". This seems to only happen on platform io because the example i used work on arduinoIDE 2.0

The problem seems to be coming from WIFI NINA generic. 

To solve this i ended up modifying the   #include <WiFiNINA_Generic.h> by <WiFiNINA.h> in WebSockets_Generic.h

PS: you also need to add lib_ignore = DueFlashStorage in platformio.ini to remove a bunch of error when adding the lib to the project

khoih-prog, 2022-10-14:
Sorry I have no issue in using Arduino IDE v1.8.19, using latest libraries' versions

```
Start WebSocketClient_NINA on SAMD_NANO_33_IOT
WebSockets_Generic v2.16.0
Used/default SPI pinout:
MOSI:11
MISO:12
SCK:13
SS:10
Please upgrade the firmware
Attempting to connect to SSID: HueNet1
SSID: HueNet1
WebSockets Client IP Address: 192.168.2.118
signal strength (RSSI):-26 dBm
WebSockets Server : 192.168.2.30
Connected to WebSockets Server @ 192.168.2.30
[WS] [WS-Client] Connect ws...
[WS] [WS-Client][connectedCb] Connected to Host:192.168.2.30, Port:8080
[WSc] Connected to url: /
[WSc] get text: Connected
[WSc] get text: message here
[WSc] get text: message here
[WSc] get text: message here
[WSc] get text: message here
```

> This seems to **only happen on platform io** because the example i used **work on arduinoIDE 2.0**

Don't you see this is strange ??? There must be something wrong in the way PIO compiles the code. I suggest you post the issue on PIO Forum and help fix the problem.

> To solve this i ended up modifying the #include <WiFiNINA_Generic.h> by <WiFiNINA.h> in WebSockets_Generic.h
> 
> PS: you also need to add lib_ignore = DueFlashStorage in platformio.ini to remove a bunch of error when adding the lib to the project

The PIO is still **immature** and not yet as smart as Arduino IDE in selecting the libraries and how to compile. Especially when dealing with **multi-platform libraries**. PIO requires better of knowledge and skills to manually modifying the settings. But this is a **good trade-off** to have a much **better developing / debugging environment**.

Anyway, this is not an issue or bug of this library and I'm closing it now.

closed 2022-10-14 by khoih-prog

aldricGit, 2022-10-14:
> Don't you see this is strange ??? There must be something wrong in the way PIO compiles the code. I suggest you post the issue on PIO Forum and help fix the problem.

Ofc I think this is strange, just wanted to let you know since you show how to install on pio that the installation is having some issue. I understand that the bug is coming from pio but since it impacts the lib I wanted to share that to you, could make you gain some time.
Anyway, thank you for the lib and the very detailled README.