khoih-prog/WebSockets_Generic #30: Network/Power failure

guotingchao, 2022-11-05:
### Describe the bug
 When the client suddenly dies/disconnects due to network/power failure in WebSockets_Generic. Since the client wont have any chance to emit a disconnect event to the server

```
Arduino IDE version: 1.8.19
Wemos-D1
OS: MacOs
Server: NodeJs exporess + SocketIO

Steps to reproduce:
1. use esp8266 example
2. upload code to borad
3. connect to server success
4. power off the broad
```

khoih-prog, 2022-11-05:
How can you expect what the library / board can do after

> client suddenly dies

or even

> power off the broad

It's your job to **restart** the WiFi / client when you get power back (if you're lucky), then restore/reconnect WiFi if WiFi lost. Check [**WiFiMulti_Generic**](https://github.com/khoih-prog/WiFiMulti_Generic) to use backup WiFi

closed 2022-11-05 by khoih-prog

guotingchao, 2022-11-06:
Hello, thank you very much for your reply.

Sorry, maybe I didn't express my question clearly. I actually use a nodejs server on the server side, using Socket.io@v4 to run the service.

And there will be both web client and esp8266 client.

The web side uses socket-io-client@v4 , the same version as the server side, and uses [transport = websocket] to restrict the link method.

1. if I disconnect the web side. For example, if I close the web page or disconnect the network of the current client, I will get [disconnect] event response on the server side immediately.

2. but when the esp8266 is disconnected from the power or network I need to delay at least 45s before I receive the disconnect response.

I would like to know if there is anything I can do to make the server side get the disconnect event first when disconnecting the network or power?

guotingchao, 2022-11-06:
> How can you expect what the library / board can do after
>
>
> or even
>
>
> It's your job to **restart** the WiFi / client when you get power back (if you're lucky), then restore/reconnect WiFi if WiFi [...]

Hello, thank you very much for your reply.

Sorry, maybe I didn't express my question clearly. I actually use a nodejs server on the server side, using Socket.io@v4 to run the service.

And there will be both web client and esp8266 client.

The web side uses socket-io-client@v4 , the same version as the server side, and uses [transport = websocket] to restrict the link method.

1. if I disconnect the web side. For example, if I close the web page or disconnect the network of the current client, I will get [disconnect] event response on the server side immediately.

2. but when the esp8266 is disconnected from the power or network I need to delay at least 45s before I receive the disconnect response.

I would like to know if there is anything I can do to make the server side get the disconnect event first when disconnecting the network or power?

khoih-prog, 2022-11-07:
HI @guotingchao 

> I would like to know if there is anything I can do to make the server side get the disconnect event first when disconnecting the network or power?

This is the **server responsibility** to use time-out to discard pending connections. The server has the option to wait how long for the client, before dropping the connection. Imagine the power outage happens randomly, unless you have a UPS, there is nothing you can do for the **dead** board to send final message or **last-will** to inform the server.

Try to research about `server timeout`. There nothing you can do on the client.