khoih-prog/WebSockets_Generic #46: multiple definition of `WebSocketsClient::*`

KSDaemon, 2023-01-23:
### Describe the bug

Hi there! First of all, thnx for your library!

I'm implementing another library that relies on this one. So to be able to use objects and methods from this library I had **exactly one** `include` of this library in my main header file.
Something like this in `mylibrary.h` 
```cpp
#include <WebSocketsClient_Generic.h>
```

I also have a `mylibrary.cpp` with again has exactly one include of its header:
```cpp
#include "mylibrary.h"
```

Now I'm making an arduino sketch (ino) where I include **only** my library header. Something like this:
```cpp
#include <mylibrary.h>
```

I'm trying to compile the sketch and am getting a bunch of errors about multiple definitions:
```
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSocketsClient::messageReceived(WSclient_t*, WSopcode_t, unsigned char*, unsigned int, bool)':
/Users/kostik/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_tree.h:2223: multiple definition of `WebSocketsClient::messageReceived(WSclient_t*, WSopcode_t, unsigned char*, unsigned int, bool)'
/sketch/mylibrary-Test.ino.cpp.o:/Users/kostik/Projects/Arduino/libraries/WebSockets_Generic/src/WebSocketsClient_Generic-Impl.h:806: first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::write(WSclient_t*, unsigned char*, unsigned int)':
mylibrary.cpp:(.text._ZN10WebSockets5writeEP10WSclient_tPhj+0x0): multiple definition of `WebSockets::write(WSclient_t*, unsigned char*, unsigned int)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets5writeEP10WSclient_tPhj+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSocketsClient::clientDisconnect(WSclient_t*)':
mylibrary.cpp:(.text._ZN16WebSocketsClient16clientDisconnectEP10WSclient_t+0x0): multiple definition of `WebSocketsClient::clientDisconnect(WSclient_t*)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN16WebSocketsClient16clientDisconnectEP10WSclient_t+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSocketsClient::clientIsConnected(WSclient_t*)':
mylibrary.cpp:(.text._ZN16WebSocketsClient17clientIsConnectedEP10WSclient_t+0x0): multiple definition of `WebSocketsClient::clientIsConnected(WSclient_t*)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN16WebSocketsClient17clientIsConnectedEP10WSclient_t+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WS_IPAddressToString(IPAddress const&)':
mylibrary.cpp:(.text._Z20WS_IPAddressToStringRK9IPAddress+0x0): multiple definition of `WS_IPAddressToString(IPAddress const&)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._Z20WS_IPAddressToStringRK9IPAddress+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::createHeader(unsigned char*, WSopcode_t, unsigned int, bool, unsigned char*, bool)':
mylibrary.cpp:(.text._ZN10WebSockets12createHeaderEPh10WSopcode_tjbS0_b+0x0): multiple definition of `WebSockets::createHeader(unsigned char*, WSopcode_t, unsigned int, bool, unsigned char*, bool)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets12createHeaderEPh10WSopcode_tjbS0_b+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::sendFrameHeader(WSclient_t*, WSopcode_t, unsigned int, bool)':
mylibrary.cpp:(.text._ZN10WebSockets15sendFrameHeaderEP10WSclient_t10WSopcode_tjb+0x0): multiple definition of `WebSockets::sendFrameHeader(WSclient_t*, WSopcode_t, unsigned int, bool)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets15sendFrameHeaderEP10WSclient_t10WSopcode_tjb+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::sendFrame(WSclient_t*, WSopcode_t, unsigned char*, unsigned int, bool, bool)':
mylibrary.cpp:(.text._ZN10WebSockets9sendFrameEP10WSclient_t10WSopcode_tPhjbb+0x0): multiple definition of `WebSockets::sendFrame(WSclient_t*, WSopcode_t, unsigned char*, unsigned int, bool, bool)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets9sendFrameEP10WSclient_t10WSopcode_tPhjbb+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::clientDisconnect(WSclient_t*, unsigned short, char*, unsigned int)':
mylibrary.cpp:(.text._ZN10WebSockets16clientDisconnectEP10WSclient_ttPcj+0x0): multiple definition of `WebSockets::clientDisconnect(WSclient_t*, unsigned short, char*, unsigned int)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets16clientDisconnectEP10WSclient_ttPcj+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::handleWebsocketPayloadCb(WSclient_t*, bool, unsigned char*)':
mylibrary.cpp:(.text._ZN10WebSockets24handleWebsocketPayloadCbEP10WSclient_tbPh+0x0): multiple definition of `WebSockets::handleWebsocketPayloadCb(WSclient_t*, bool, unsigned char*)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets24handleWebsocketPayloadCbEP10WSclient_tbPh+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::headerDone(WSclient_t*)':
mylibrary.cpp:(.text._ZN10WebSockets10headerDoneEP10WSclient_t+0x0): multiple definition of `WebSockets::headerDone(WSclient_t*)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets10headerDoneEP10WSclient_t+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::base64_encode(unsigned char*, unsigned int)':
mylibrary.cpp:(.text._ZN10WebSockets13base64_encodeEPhj+0x0): multiple definition of `WebSockets::base64_encode(unsigned char*, unsigned int)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets13base64_encodeEPhj+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::acceptKey(String&)':
mylibrary.cpp:(.text._ZN10WebSockets9acceptKeyER6String+0x0): multiple definition of `WebSockets::acceptKey(String&)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets9acceptKeyER6String+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::write(WSclient_t*, char const*)':
mylibrary.cpp:(.text._ZN10WebSockets5writeEP10WSclient_tPKc+0x0): multiple definition of `WebSockets::write(WSclient_t*, char const*)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets5writeEP10WSclient_tPKc+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::enableHeartbeat(WSclient_t*, unsigned int const&, unsigned int const&, unsigned char const&)':
mylibrary.cpp:(.text._ZN10WebSockets15enableHeartbeatEP10WSclient_tRKjS3_RKh+0x0): multiple definition of `WebSockets::enableHeartbeat(WSclient_t*, unsigned int const&, unsigned int const&, unsigned char const&)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets15enableHeartbeatEP10WSclient_tRKjS3_RKh+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::handleHBTimeout(WSclient_t*)':
mylibrary.cpp:(.text._ZN10WebSockets15handleHBTimeoutEP10WSclient_t+0x0): multiple definition of `WebSockets::handleHBTimeout(WSclient_t*)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets15handleHBTimeoutEP10WSclient_t+0x0): first defined here
[... 135 lines cut ...]
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::handleWebsocketCb(WSclient_t*)':
mylibrary.cpp:(.text._ZN10WebSockets17handleWebsocketCbEP10WSclient_t+0x0): multiple definition of `WebSockets::handleWebsocketCb(WSclient_t*)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets17handleWebsocketCbEP10WSclient_t+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSockets::handleWebsocket(WSclient_t*)':
mylibrary.cpp:(.text._ZN10WebSockets15handleWebsocketEP10WSclient_t+0x0): multiple definition of `WebSockets::handleWebsocket(WSclient_t*)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN10WebSockets15handleWebsocketEP10WSclient_t+0x0): first defined here
/libraries/mylibrary/mylibrary.cpp.o: In function `WebSocketsClient::handleClientData()':
mylibrary.cpp:(.text._ZN16WebSocketsClient16handleClientDataEv+0x0): multiple definition of `WebSocketsClient::handleClientData()'
[... cut ...]
mylibrary.cpp:(.text._ZN16WebSocketsClient7onEventESt8functionIFv8WStype_tPhjEE+0x0): multiple definition of `WebSocketsClient::onEvent(std::function<void (WStype_t, unsigned char*, unsigned int)>)'
/sketch/mylibrary-Test.ino.cpp.o:mylibrary-Test.ino.cpp:(.text._ZN16WebSocketsClient7onEventESt8functionIFv8WStype_tPhjEE+0x0): first defined here
collect2: error: ld returned 1 exit status
```

Actually compilation succeeds, but the linking stage — not :(

I am using ESP32 `DOIT ESP32DEVKIT V1` board but i suppose that doesn't matter.
Test examples of the library works as expected.

So why is it happening? Am i doing something wrong? Can you advice? 
Thnx!

KSDaemon, 2023-01-23:
Btw, I saw another one your lib [WebSockets2_Generic](https://github.com/khoih-prog/WebSockets2_Generic). What's the principal difference between them? As I see it uses the same approach. So would I experience the same problem if I switch to that one? :)

khoih-prog, 2023-01-25:
HI @KSDaemon 

The real reason is that these libraries use the new **h-only implementation**. Check [**HOWTO Fix Multiple Definitions Linker Error**](https://github.com/khoih-prog/AsyncHTTPRequest_Generic#howto-fix-multiple-definitions-linker-error) and many issues / discussions I don't remember now in my libraries.

The solution is either  
- **using h-only project or**
- **to include the libraries' h files only once, in main `.ino` or `main()`**

--- 

### A)  h-only project

#### 1. main `MultiDefError.ino`

```cpp
#include "mylibrary.h"

void setup() 
{
  // put your setup code here, to run once:
}

void loop() 
{
  // put your main code here, to run repeatedly:
}
```

#### 2. `mylibrary.h`

```cpp
#ifndef MY_LIB_H
#define MY_LIB_H

// OK if using h-only project
#include <WebSocketsClient_Generic.h>

void testFunc();

#include "mylibrary_Impl.h"

#endif
```

#### 3. `mylibrary_Impl.h`

```cpp
#ifndef MY_LIB_IMPL_H
#define MY_LIB_IMPL_H

void testFunc()
{
}

#endif
```

---
---

### B)  cpp project

#### 1. main `MultiDefError_cpp.ino`

```cpp
// Include only in main .ino or main() to avoid multi-def error if using mixed cpp
#include <WebSocketsClient_Generic.h>

#include "mylibrary.h"

void setup() 
{
  // put your setup code here, to run once:
}

void loop() 
{
  // put your main code here, to run repeatedly:
}
```

#### 2. `mylibrary.h`

```cpp
#ifndef MY_LIB_H
#define MY_LIB_H

void testFunc();

#endif
```

#### 3. `mylibrary_Impl.h`

```cpp
#include "mylibrary.h"

void testFunc()
{
}
```

---
---

Using [**WebSockets2_Generic**](https://github.com/khoih-prog/WebSockets2_Generic) library the same way

Just a simple and quick explanation, and **you have to dig in and figure out how and what to do**. I have no more time here.

Good Luck,

closed 2023-01-25 by khoih-prog