khoih-prog/WebSockets_Generic #49: Ability to receive the Response HTTP Headers

KSDaemon, 2023-01-26:
I see that HTTP headers are populated somewhere in the deep of the library. But from client's side they are not available. Most of them are just technical. But one of the interests is `sec-websocket-protocol`. 
Because as described in RFC 6455 client can send a few desired protocols and the server may choose one of them. But client can not get information on what protocol was chosen by the server.

So it would be really helpful to have a kind of `.getHeaders()` or similar public method on the WebSocketsClient to get this information or it can be passed back to `onWsEvent` callback alongside with type, payload and length.

khoih-prog, 2023-01-26:
Hi @KSDaemon 

This library is just a port of [**arduinoWebSockets**](https://github.com/Links2004/arduinoWebSockets) library to provide supports to many other boards besides ESP32/ESP8266, and I don't have time and interest in maintaining this library anymore, unless there is some real bug affecting the normal usage of those new boards.

I suggest you post any general issue in original [**arduinoWebSockets**](https://github.com/Links2004/arduinoWebSockets) library.

Good Luck,

closed 2023-01-26 by khoih-prog

KSDaemon, 2023-01-28:
Hi @khoih-prog I see. Thnx for the reply! I'll ask my question in the arduinoWebSockets repo!

khoih-prog, 2023-01-28:
Good to know you'll help. 

Please post the link of your issue so that we, me including, can benefit from that.

KSDaemon, 2023-01-28:
Of course I'll link it!